sn4k3/UVtools #1: [FEATURE REQUEST] Erode bottom layers only

RebelKeithy, 2020-05-30:
**Is your feature request related to a problem? Please describe.**
When printing on SLA printers the bottom layers must be over exposed to adhere to the build plate. This causes these layers to expand.

**Describe the solution you'd like**
To prevent this expansion from making the bottom layers larger than the normal layers, I would like to be able to erode just the bottom layers.

**Additional context**
Adding a checkbox to the erode menu to have it operate on just the bottom layers to be perfect.

sn4k3, 2020-05-30:
The feature you request is already being implemented, will be avaliable on next build as many other news ;)

RebelKeithy, 2020-05-30:
That's awesome. This is a great app for filling in features missing in slicers. In the meantime I've managed to hardcode it to only erode the first 6 layers 😄

sn4k3, 2020-05-30:
Here the last dev with that feature: https://easyupload.io/3ffnmh

RebelKeithy, 2020-05-30:
Just used to print some dungeon tiles. It works great. Thanks!

closed 2020-06-01 by sn4k3

sn4k3, 2020-06-01:
It's now possible to fade erode, decreasing the iteration over layer height for better overexposure solution