sn4k3/UVtools #2: [BUG] Conversion of .sl1 to .cbddlp results in garbled, 'diagonal scanline' output

tofof, 2020-06-05:
## Description
The .cbddlp output is garbled.  Seems like the output you'd get if you serialized the layer's pixels in row order, but then reassembled them in column order (where row and column have different dimensions). The pixels from the rightmost shape end up on top, and leftmost on bottom.

## To Reproduce
Steps to reproduce the behavior:
1. Open `1_sliced-by-prusaslicer.sl1` [testcase.zip](https://github.com/sn4k3/PrusaSL1Viewer/files/4733545/testcase.zip) in PrusaSL1Viewer
2. Click on `File` -> `Convert To` -> `Chitubox (.cbddlp ....)` and save. 
My output for this step is the file `2_coverted-by-PrusaSL1Viewer.cbddlp`.
3. Open the saved file and view unexpected result 

## Expected Behavior
Pixels in the Chitubox output file are placed identically as in the SL1 input file.

## Screenshots
[image]
[image]

## System Information
 - OS: Windows 10
 - PrusaSL1Viewer: 0.4.2.1

sn4k3, 2020-06-05:
You can't slice on PE using Prusa SL1 printer, you must use one of my pre configured printers or else you will get this problem since prusa printer is not recognized for .ctb or other formats. Please follow readme point: "Install and configure printer under PrusaSlicer" and use your printer name to slice on PE

tofof, 2020-06-05:
PS - really like the tool! 
Dramatically better automatic island repair than the photonsters Photon File Validator. In particular, the islands on layer 116-117 in that file would've been a bear to do in the other tool. I also appreciate that you can actually drag the mouse to paint pixels in pixel editing mode instead of having to click each individual pixel. 
The one place where the other tool wins right now is the 'difference' pixels. It's a lot easier to see yellow pixels that have nothing beneath them than it is to look at the darker greys in your tool.

tofof, 2020-06-05:
> You can't slice on PE using Prusa SL1 printer, you must use one of my pre configured printers or else you will get this problem since prusa printer is not recognized for .ctb or other formats. [...]

Ok, that's my bad for not paying attention to the readme.

sn4k3, 2020-06-05:
> Ok, that's my bad for not paying attention to the readme.
No problem, in fact i can do any kind of auto detection and fix this but that would cause another problem, which are what you did. Imagine if the file converted ok, then you go print and them it fail cause wrong printer parameters, PE is to be used like any other slicer, if you go print on a epax then you select epax printer to slice and not other wrong printer profile, now imagine the problem if you slice using epax x10 and go print under a x1. So i leave that way to prevent users using SL1 for any dangerous print. I think no one will attempt to print when they see that crazy lines xD

sn4k3, 2020-06-05:
> PS - really like the tool!
> Dramatically better automatic island repair than the photonsters Photon File Validator. In particular, the islands on layer 116-117 in that file would've been a bear to do [...]

Thank you for your input, for any improvement or ideia you can open a feature request here.
PS: I always recomend 1 pass layer fix even if don't have islands, it can fix several problems ;)

If you fell this issue is fixed please consider closing it
Regards

tofof, 2020-06-05:
Makes sense. 

I suggest that you might want to automatically call the 'install printers' dialog the first time the program is run, along with info that selecting the appropriate printer in PrusaSlicer is mandatory for file format conversions in this tool.

And yes, I just confirmed that fixes things. Thanks!

closed 2020-06-05 by tofof