sn4k3/UVtools #3: [FEATURE REQUEST] Fill Cavities

RebelKeithy, 2020-06-05:
**Is your feature request related to a problem? Please describe.**
When hollowing a model in the slicer, sometimes there are small cavities that can't use drain holes.

**Describe the solution you'd like**
The program should be able to go through layer by layer and find cavities that don't have holes, and fill these cavities with white pixels. 

**Describe alternatives you've considered**
No easy alternatives to this except leave the cavities filled with liquid resin

**Additional context**
I realize this isn't a simple feature request. My initial thoughts on how to do this would be:
1: Flag all black pixels on the boarder of the 3d space
2: For each black pixel that is not flagged
3:     If it is adjacent to a flagged pixel, flag it
4: If a pixel was flagged in the last step, go to 2
5: All black pixels that are not flags are set to white

sn4k3, 2020-06-05:
I don't think i full understand the problem, can you send a image sample? Note that i'm not used to resin printing, i never own one and real problems on resin printing are all new to me.
The solution you indicate can be solved using the mutators already in app

RebelKeithy, 2020-06-05:
Here's an example of the problem. I've hollowed this model with chitubox and have a cavity here that doesn't have a drain hole. When printed it will be filled with liquid resin, so if anything happens to crack the print there could be hazardous resin leaking out.
[image]
[image]

sn4k3, 2020-06-05:
As i understand every drain hole must be done by user & inspection, chitubox hallow doesn't generate them right? On that case you don't want do a drain hole on model but just fill that area correct?
I can start thinking over that, a hallow area detector...

RebelKeithy, 2020-06-05:
Yes, that area is so small (about 1.5mm wide) there's no place to make a drain hole.

sn4k3, 2020-06-05:
I will take that into consideration and good feature to have.
Today i will release a new version which fix cbt and cbddlp file save, if you were saving files for that formats with AA > than 1 please discard them and redo with new version

sn4k3, 2020-06-05:
Also consider using PrusaSlicer instead o chitubox, way better...

sn4k3, 2020-06-19:
https://github.com/sn4k3/UVtools/releases/tag/v0.5.2

closed 2020-06-19 by sn4k3