sn4k3/UVtools #4: [BUG] File with "Printer absolute correction" other than zero won't open

b-fitzpatrick, 2020-06-25:
**Describe the bug**
When attempting to load a file from PrusaSlicer where the "Printer absolute correction" setting is not zero, it never loads. The "Loading..." dialog just keeps waiting.

**To Reproduce**
Steps to reproduce the behavior:
1. On the "Printer Settings" tab of PrusaSlicer, set "Printer absolute correction:" to -0.05.
2. Slice and export.
3. Open the exported file with UVtools
4. It never opens.

**Expected behavior**
I expect the file to open and for the correction to be applied.

**Desktop (please complete the following information):**
 - OS: Windows 10 Pro v1803
 
I want to decrease the size of walls, similar to "Erode" in UVtools, but I would like to do so by a small amount to calibrate the parts from the printer. For example, I printed a 10mm x 10 mm open-ended box with 1 mm walls. The walls measure around 1.1 mm thick, and the box width measures around 10.1 mm.

Looking at the .png files in the .sl1 output file, it appears that "Printer absolute correction" is working as expected.

In "SL1File.cs", variable "AbsoluteCorrection" is typed as a byte, but it should probably be a float. I confirmed that setting "Printer absolute correction" to 1 in PrusaSlicer allows the file to open, and the layer images are inflated by 1 mm.

sn4k3, 2020-06-25:
You are right, byte value will trigger error when try to parse a float.
Will be fixed on next release.
Meanwhile you can do the same on UVtools. Use Resize function and resize by 99.95%

b-fitzpatrick, 2020-06-25:
Thanks. I don't think resize is the same thing. I need to remove 0.05 mm from each outside surface, similar to "Erode" in UVtools.

In the meantime, I'll just edit the prusaslicer.ini file, since it doesn't seem like that value is used for anything. FYI, the same issue exists with GammaCorrection.

sn4k3, 2020-06-25:
After slice that value is just informative, you can't change .ini and expect that will print with correction.
Here the v0.6 beta with that fixed: https://easyupload.io/vk0hvz
Try and report back.
I fixed a bunch of bytes to floats, inc GammaCorrection

b-fitzpatrick, 2020-06-28:
I've confirmed that the fix works. Thanks.
FYI, I was changing the .ini file that's contained inside the .sl1 file, to avoid the bug. The corrections were already made during slicing, so the print was correct.

sn4k3, 2020-06-28:
Can you give the Bene specs so i can add it?

b-fitzpatrick, 2020-06-28:
I don't know anything about Bene specs. I'm guessing this comment was intended for the other open issue.

sn4k3, 2020-06-28:
You are right, sorry for confusion

closed 2020-06-29 by sn4k3