sn4k3/UVtools #5: Nova3D Elfin Machine settings and cws file [BUG]

Nume1977, 2020-06-26:
1) Elfin Machine profile settings are not correct:
Display Width : 131mm /  Height : 73mm
Screen X: 2531 / Y: 1410

2) The exported cws file reports corrupted by the printer, i think it is because of the file naming inside the zip.

AFAIK, exported files inside the zip (cws) must all be named 'slice':
slice.conf
slice.gcode
slice00000.png
slice00001.png
etc..
(note: slice(5 digits).png)

Regards,
Rui

sn4k3, 2020-06-26:
Is dam hard to get the true values, many websites have diferent values regarding volume as resolutions, i will check that!

About names i don't think thats the problem, see:
[image]

sn4k3, 2020-06-26:
Please try this build and r eport back: https://easyupload.io/9za8iv

sn4k3, 2020-06-26:
About resolution and display every website says that:
[image]

Where you get your values?

Nume1977, 2020-06-26:
> Where you get your values?

From the facebook community and private Elfin Mono LCD screen group, the real colour screen size is the one i posted above.
These values are also on the Chitubox cws export plugin, and they work correctly.

About the names, maybe they need to be all lower case. I remember i had some trouble with the Chitubox plugin and it was related to the filenames. I will test the new version and report back in the meantime.

sn4k3, 2020-06-26:
Is elfin mono lcd? or they have both versions?

sn4k3, 2020-06-26:
I tested the Chitubox plugin, but it's not native .cws format, is just the zip format renamed to cws...
Thats the same you convert to Zip under UVtools or chitubox

Nume1977, 2020-06-26:
> 
> 
> Is elfin mono lcd? or they have both versions?

There is going to be a new mono screen version (Elfin and probably Bene) and there might be an upgrade kit for Elfin1.
Early tests point to a 2s layer cure time ;)
But the Elfin1 as a serious problem with screen over heat, i am making some mods that will be public, so the screens can last more than a few months!

About the cws file format, i've figured out what the problem is :), and it prints now .

It's the header on the gcode file, it must be like this:

`;estimatedPrintTime:1487.7
;volume:3.32817
;resin:normal
;weight:3.66099
;price:0.133127
;layerHeight:0.05
;resolutionX:1410
;resolutionY:2531
;machineX:73
;machineY:131
;machineZ:151
;projectType:LCD_mirror
;normalExposureTime:6.5
;bottomLayExposureTime:30
;bottomLayerExposureTime:30
;normalDropSpeed:180
;normalLayerLiftHeight:4
;zSlowUpDistance:0
;normalLayerLiftSpeed:120
;bottomLayCount:3
;bottomLayerCount:3
;mirror:1
;totalLayer:276
;bottomLayerLiftHeight:6
;bottomLayerLiftSpeed:120
;bottomLightOffTime:0
;lightOffTime:0
;Layer Thickness = 0.05
;Number of Slices = 276
;Z Lift Feed Rate = 120.0
;Lift Distance = 4.0`

Instead of : 
`;(Pix per mm X             = 19.324)
;(Pix per mm Y             = 19.324)
;(X Resolution             = 2531)
;(Y Resolution             = 1410)
;(Layer Thickness          = 0.05)
;(Layer Time               = 6000)
;(Render Outlines          = False)
;(Outline Width Inset      = 2)
;(Outline Width Outset     = 0)
;(Bottom Layers Time       = 30000)
;(Number of Bottom Layers  = 3)
;(Blanking Layer Time      = 0)
;(BuildDirection           = Bottom_Up)
;(Lift Distance            = 4)
;(Slide/Tilt Value         = 0)
;(Use Mainlift GCode Tab   = False)
;(Anti Aliasing            = True)
;(Anti Aliasing Value      = 4)
;(Z Lift Feed Rate         = 120)
;(Z Bottom Lift Feed Rate  = 120)
;(Z Lift Retract Rate      = 120)
;(Flip X                   = True)
;(Flip Y                   = True)
;(Number of Slices         = 276)
;(Platform X Size          = 0)
;(Platform Y Size          = 0)
;(Platform Z Size          = 0)
;(Max X Feedrate           = 200)
;(Max Y Feedrate           = 200)
;(Max Z Feedrate           = 200)
;(Machine Type             = UV_LCD)
;(Bottom Layer Light PWM   = 255)
;(Layer Light PWM          = 255)`

sn4k3, 2020-06-26:
There is something not matching here...
Isn't NovaMaker the default slicer for Elfin? Because i use that to produce the files and parse them with uvtool. NovaMaker produce files like i am
[image]

sn4k3, 2020-06-26:
Try this build: https://easyupload.io/icd50j
Must fix the problem

sn4k3, 2020-06-28:
Can you give the Bene specs so i can add it as a printer?

closed 2020-06-29 by sn4k3