sn4k3/UVtools #6: [BUG] Unparsable for errors sl1 file

Nume1977, 2020-06-26:
After changing the Nova3D Elfin printer settings, i resaved the sl1 file, but now it is unparsable for errors and UVTools is "stuck" (can't change any settings on this file, i.e. number of layers).

**To Reproduce**
1. Open the attached file
2. Goto Issues tab
3. "Error while trying to parse errors" pops up

[Capa_quadrada_DMG.zip](https://github.com/sn4k3/UVtools/files/4837480/Capa_quadrada_DMG.zip)

sn4k3, 2020-06-26:
It seens you change resolution and image convertion fails without a reason... Anyway this is fixed with new version (not released yet), but you can grab it here: https://easyupload.io/o7z3d4
It will solve your problem and others more

Nume1977, 2020-06-26:
Fixed, now parses the file!

closed 2020-06-26 by Nume1977