sn4k3/UVtools #7: [FEATURE REQUEST] - Paint tools - Flood fill, box draw, pencil size

X3msnake, 2020-06-29:
One thing that Validator has that is very usefull is the box paint method.
Another thing that would be very usefull would be a flood fill method.

- Box fill
- Flood fill
- Pencil tip size
- Text generator

sn4k3, 2020-06-29:
Improve pixel edit function is on my list for the future release, including brush size and shape.
Can you explain what can be expected from Box fill and Flood fill?

X3msnake, 2020-06-29:
Flood Fill
[image]

X3msnake, 2020-06-29:
[image]

sn4k3, 2020-06-29:
Thank you for details, it will be possible.
Regards

sn4k3, 2020-08-11:
Implemented

closed 2020-08-11 by sn4k3