sn4k3/UVtools #11: [FEATURE REQUEST] Remember last selected file type

RebelKeithy, 2020-07-10:
**Is your feature request related to a problem? Please describe.**
Every time I use the program it's for phz files, and every time I open a file I have to select phz from the dropdown menu. Most people will only be working with one type of file and will have the same issue.

**Describe the solution you'd like**
Program should remember last selected filetype and automatically select that one.

**Additional context**
Not a huge issue, but would be nice.

sn4k3, 2020-07-10:
That is implement since v0.6.1.0, you can choose default file extension at settings. Please check

RebelKeithy, 2020-07-10:
I knew I should update before making a request. Thanks

closed 2020-07-10 by RebelKeithy