sn4k3/UVtools #12: [BUG] Issues with custom values in printer notes

b-fitzpatrick, 2020-07-17:
In the LookupCustomValues method in SL1File.cs, it searches for the custom values by string with an underscore added (i.e., "LiftHeight_"). This fails, as it finds "BottomLiftHeight_" and uses that value. The same occurs for "LiftSpeed_".

Also, I haven't found the code yet, but when importing the printer notes, it appears to truncate decimals. For example, I have "BottomLiftHeight_8.5" in the printer notes, but in the PrinterNotes string, the ".5" is not present.

sn4k3, 2020-07-17:
This affects which format? you trying convert SL1 to what format?

b-fitzpatrick, 2020-07-17:
I'm converting to ChituBox format for an Elegoo Mars printer.

sn4k3, 2020-07-17:
Ok i found the problem. Will be fixed on next release

sn4k3, 2020-07-17:
[image]
Fixed

b-fitzpatrick, 2020-07-17:
Thanks! I'm compiling from source, so I'll pick it up when you commit.

sn4k3, 2020-07-17:
> Thanks! I'm compiling from source, so I'll pick it up when you commit.

Here's the fix code if you want to compile it right now:

`public T LookupCustomValue<T>(string name, T defaultValue, bool existsOnly = false)
        {
            if (string.IsNullOrEmpty(PrinterSettings.PrinterNotes)) return defaultValue;
            string result = string.Empty;
            if(!existsOnly)
                name += '_';

            var lines = PrinterSettings.PrinterNotes.Split(new[] { "\\r\\n", "\\r", "\\n" },
                StringSplitOptions.RemoveEmptyEntries);

            for (var i = 0; i < lines.Length; i++)
            {
                var line = lines[i].Trim();
                if (!line.StartsWith(name)) continue;
                if (existsOnly || line == name) return "true".Convert<T>();
                var value = line.Remove(0, name.Length);
                for (int x = 0; x < value.Length; x++)
                {
                    char c = value[x];
                    if (!char.IsLetterOrDigit(c) && c != '.')
                    {
                        break;
                    }

                    result += c;
                }
            }

            return string.IsNullOrWhiteSpace(result) ? defaultValue : result.Convert<T>();
        }`

b-fitzpatrick, 2020-07-17:
That fixes the first issue. The second issue obviously still exists.

sn4k3, 2020-07-17:
> That fixes the first issue. The second issue obviously still exists.

Can you better describe it?
I test float values and they work ok to me

b-fitzpatrick, 2020-07-17:
Sorry, my bad. It works properly. I had removed the 0.5 in my test .sl1 file.

closed 2020-07-17 by b-fitzpatrick