sn4k3/UVtools #13: [BUG] Resin trap reported on exterior of model

tofof, 2020-07-28:
**Describe the bug**
Resin trap reported that's actually completely open to the exterior of the model.

**To Reproduce**
Steps to reproduce the behavior:

1. Issues
2. Resin trap 4, which seems to be only the left (y=1300) basin, judging from what the repair tool changes
3. Observe that the 'trapped' area is actually open to the sky, ie all layers above that area to the top are completely empty in that region

**Expected behavior**
Resin trap is only reported for 1-3, which are real. 
Left basin (y=1300) acts like right basin (y=1264).

**Files**
[DOG_STL_2.zip](https://github.com/sn4k3/UVtools/files/4985455/DOG_STL_2.zip)

**Desktop (please complete the following information):**
 - OS: Win 10 Pro x64 1803
 - UVtools 0.6.3.3

sn4k3, 2020-07-28:
Yeah i just confirm it, need to review the heavy code from resin traps.

sn4k3, 2021-03-19:
"Change resin trap search from parallel to sync to prevent fake detections and missing joints at cost of speed."
Need to rewrite resin traps soon or latter...

closed 2021-03-19 by sn4k3