sn4k3/UVtools #14: [DOCUMENTATION] Menu item changes not reflected in readme

tofof, 2020-07-28:
Very minor.

**Describe the bug**
Readme currently says: "Under Menu click -> About -> Install profiles into PrusaSlicer"

**Expected behavior**
"Help -> Install profiles into PrusaSlicer"

sn4k3, 2020-07-28:
Thanks, will be fixed on next release

tofof, 2020-07-28:
Keep up the great work! This project is really progressing nicely.

Nice explanations in the mutate tool windows, particularly the pictures. If you can, pictures on all of them would be even better.
I'm still not sure what PyrDownUp does, for example, (particularly the bit about 'specified filter' since there doesn't seem to be a way to specify one) and it doesn't have pictures.

sn4k3, 2020-07-28:
As single developer not easy to carry this and polish everything since now everything require a good amount of time to do, if you want, fell free to send documentation or images to complete the program.
PyrDownUp is easy spoted, it add a huge blur, or over exagerated AA, i think no one tried this out, but this have potential or not to create a very smooth edges.
true explanation: https://docs.opencv.org/2.4/doc/tutorials/imgproc/pyramids/pyramids.html

closed 2020-07-28 by sn4k3