sn4k3/UVtools #15: [BUG?] modifying file and resaving caused timeout (CTB)

X3msnake, 2020-07-30:
[image]

File was edited the following way.
- Removed small holes
- Pixel dimmed checker the base layers
- Added Big Blur to some layers of the supports to see the effcet

sn4k3, 2020-07-30:
Timeout may mean it took to much time parsing file and quit after a while.
Can re-try with other drive? If does the same try to remake file from original, if same happens start to remove steps: remove blur and retry.

X3msnake, 2020-07-30:
Does the same thing i think it is the pixel dimming that is blowing up the processor on the phenom

Trying the same file but editing the pixel dimming on the base layers

https://drive.google.com/file/d/19-JUodeY33lta7l98urzUhtc3W_1XdEm/view?usp=sharing

sn4k3, 2020-07-30:
Makes sense, the way RLE is, that pattern is increasing the processing for sure. Compare filesize original vs Pixel Dimming

sn4k3, 2020-07-30:
Yep a file with 13.7MB went 225MB after pixel dimming, nothing i can do here, use a more sparse pattern. Dam RLEs!
If you can use ZIP file, much better

X3msnake, 2020-07-31:
This is a hardware issue, closing it. thanks Master

closed 2020-07-31 by X3msnake