sn4k3/UVtools #16: Noise Removal introduces Islands

byancey, 2020-08-08:
I'm not sure if this is a bug, or just my lack of understanding of the settings, so submitting this first as a question rather than an issue.  

I'm  not certain exactly what the purpose of noise removal iterations is, but I'm finding that if it is set this value to anything other than 0 (default is 1), it has a tendency to completely remove the tops of some of my light supports, resulting in the introduction of islands that otherwise were not islands.  I can step layer-by-layer before and after performing a repair and observe this happening.  My guess is that due to my light supports being very small, they are treated as noise for some reason once the tips get below a certain size, and are removed as a result, even though the tips are supported from below.  No idea if this is the intended function of this setting, or just a bug. 

For now, I've just changed the default value for noise iterations to 0 so that I don't have to remember to set it to 0 manually each time I run repair.  What exactly is the purpose of the noise remove iterations, and should I be leaving this enabled and tweaking some other setting to address my issue of fine supports being clipped off?

Thanks!!

sn4k3, 2020-08-09:
Noise removal eliminates tiny groups of alone pixels, which in some cases can remove required pixels.
Iteration = 1 will erode 1px from mass perimeter, after it dilate by 1px to restore image size, but eroded groups which turn into all blacks will not be dilated since they are erased. The more iterations more it will kill the noise/details.

Example:

Take this square 3x3
* * *
* * *
* * *

Run noise removal - iteration = 1:

Original / Erode / Dilate
* * *                         * * *
* * *    =>     *   => * * *
* * *                         * * *

in this case nothing happen to the object but if square was 2 x  2 for example the result be be erased objected.

In the end is just what you think, very small details will be erased.
I will set default to 0 in next version, i already should done that as many people print more small details than big parts

byancey, 2020-08-09:
Thanks for the explanation.  I do tend to print a lot of very small miniatures with fine details, so this probably isn't a feature I'm very likely to use.  I've got it defaulting to zero in my settings, so I'm covered here, but probably not a bad idea to set the software default to 0 as well.

Now that I understand this better, I have a follow-up question.  As I was troubleshooting this, I noticed that I can turn gap closing off (iterations=0), or I can turn noise removal off (iterations=0), but if I try to turn both of them off by setting both iteration values to 0, I get an error "Any of opening and closing iterations must be non 0."

I think I understand what gap closing is doing, and with your explanation I now understand what noise removal is doing.  These are both nice feature enhancements to assist with island removal, but they appear to be independent of the actual island removal functionality itself.  Is there any reason not to be able to turn both of these features off and just rely on UVTools to remove existing islands (basically default to the same functionality found in photon validator).

Oh, great tool BTW.  I just recently switched to a new large format SLA printer which only support CTB format.   I was previously using photon validator, but it does not support the CTB format, so I was glad to  find an alternative that did support CTB.  I'm also finding some of the extra layer editing tools provided by UVTools to be extremely useful for manual layer corrections.

Thanks!!

sn4k3, 2020-08-09:
Just un-check: Repair Layers and issues, gap closing and noise removal will only run if that check box is ticked, but yeah i can add extra code to auto disable it when both are 0 and not trigger such error.
Carefull with island removal, don't remove big masses

closed 2020-08-11 by sn4k3