sn4k3/UVtools #17: [BUG] CWS Export

AlexChanson, 2020-08-09:
I just discovered this after reverse engineering the CWS format used by the Nova3D ELFIN, I'd love to use it since prusa slicer seems real good at auto supports (I'm lazy) but the G-Code you generate seems to be off.
The delays should be expressed in ms not seconds (at least in the 'original' NovaMaker files).

An example slice (0.035 mm profile):
`;<Slice> 23
M106 S255
;<Delay> 6
M106 S0
;<Slice> Blank
G1 Z4 F120
G1 Z-3.97 F120 -> other slices have 3.96 looks like a rounding problem to me
;<Delay> 6`

The 0.035 mm settings leads to some slices of 0.04 and others of 0.03 but never 0.035 this also seems like a bug. 

I'd fix it and send a pull request if I was familiar with a C# workflow but all I can offer is my small documentation on the format and some example files working on my printer I build with a python script.

The aforementioned documentation off the original G-Code https://github.com/AlexChanson/ELFIN_Z-Suite_Comaptibility/blob/master/format.md and a calibration cube I printed https://drive.google.com/file/d/1ZeBp4iQf6iu2IWkuXzVdufT12FI26LOC/view?usp=sharing seems to be 5% too big along the Y axis but otherwise printed fine.

sn4k3, 2020-08-09:
Yes, i forget to change to ms when implented per layer settings. Will be fixed on next release

AlexChanson, 2020-08-10:
Great , I quickly went over UVtools.Core/FileFormats/CWSFile.cs and the UpdateGCode method 
It seems after the lift sequence you add the exposure time as the delay `GCode.AppendLine($"{GCodeKeywordDelay} {layer.ExposureTime}");` (L633) this works if all settings are default (feed, lift distance, extra blank time) but as soon as you lower the lift speed it's too short and the exposition starts before the platform is lowered to the target height.

You might want to compute this with something along those lines : `max(<extra_blank_time>, 500) + (<lift_distance>/(<lift_feed_rate>/60) + <lift_distance>/(<lower_feed_rate>/60))*1000` here I assume distance is in mm and the feed rates in mm/minute and add an extra 500 ms for good measure (could probably be a bit less). 

The bottom layers are completely hard-codded in NovaMaker with 20mm/m feed and an extra 10s of blank time that adds up to a 34s delay for the nominal lift distance of 4mm. This is probably for avoiding any overheating.

sn4k3, 2020-08-10:
Ok i've added that too. Math is giving 6100 delay over normal layers with a lift of 7mm @ 150mm/s (same for retract)
Is that 6100 value ok in that case? extra_blank_time was 0 maxed to 500.
what default value nova use for extra_blank_time?

AlexChanson, 2020-08-10:
I was confused for the better part of an afternoon with this no waiting behavior, default extra_blank_time is 2s and 10s for bottom layers and this is a minimum can't go below probably to do with heat.
Your exemple seems good : 5600 ms of travel + 500 extra adds up to 6100

I'm also printing some calibration cubes to check if NovaMaker is accurately translating dimensions if so should be able to compute the actual screen size and pixel per millimetres values because everywhere I look I get different numbers.

sn4k3, 2020-08-10:
Yes resolutions are a mess, diferent values almost each information. But the actual right values are the one you see on link you sent for config zsuite, same values uvtools uses.
Today i will send the patch for download

AlexChanson, 2020-08-10:
I'm looking at less than 0.5% error on the 20mm cube across X and Y with Z a bit worse around 1.5% but I printed it without support, Probably seeing more resin shrinkage than anything else.

sn4k3, 2020-08-10:
You always need to find the shrinkage factor, then use uvtools to correct using resize by the factor

sn4k3, 2020-08-11:
Implemented

closed 2020-08-11 by sn4k3

AlexChanson, 2020-08-11:
[image]
Everything seems to work ! (Except my sandpaper)

sn4k3, 2020-08-11:
Thanks for confirming! Amazing cube btw