sn4k3/UVtools #18: [BUG] Model is stretched after exporting to .ctb file

GuillaumeRives, 2020-08-11:
**Describe the bug**
Model's preview is stretched after exporting it to .ctb.
It may still works since the layer preview seems to be correct. I'll try to print it anyway.

**To Reproduce**
Steps to reproduce the behavior:
1. Import sl1 file to UVtools
2. Search for issues
3. Repair ResinTrap issues
4. Export to .ctb file
5. Monitor the export on Chitubox

**Expected behavior**
Model's preview dimensions should remain the same after exports.

**Screenshots**
[image]

**Files**
[files.zip](https://github.com/sn4k3/UVtools/files/5056316/files.zip)

**Desktop (please complete the following information):**
 - OS: Windows 10 x64 v2004

sn4k3, 2020-08-11:
Yep that's normal, i don't know why but this always happen. Whats matter is the masked image, ignore the preview, it's all good :)

GuillaumeRives, 2020-08-11:
Yep, I saw this. Masked images are correct so... It's more of a comfort issue than a real one ;)

closed 2020-08-12 by sn4k3