sn4k3/UVtools #19: [BUG] Elegoo Mars Pro profile not working

GuillaumeRives, 2020-08-11:
**Describe the bug**
I don't know if it's a bug or a my misunderstanding but I have an Elegoo Mars Pro so I imported the Elegoo Mars 2 Pro profile into PrusaSlicer. When I try to print a model with this profile, my printer tell me the screen dimensions are mismatching. I tried with the Elegoo Mars profile and it works since those two printers have the same screen dimensions.

**Expected behavior**
Is this Elegoo Mars 2 Pro profile intended to cover the Elegoo Mars Pro or a new Elegoo Mars printer I'm still unaware of ?
To avoid misunderstanding, I suggest to add a new profile import choice named Elegoo Mars Pro with its real caracteristics.
[image]

**Desktop (please complete the following information):**
 - OS: Windows 10 x64 v2004

sn4k3, 2020-08-11:
You have done a a new printer into PrusaSlicer? If so you need implement some stuff into Notes. 
The easy way is use UVtools printers: go to UVtools Help -> Install Printers into PrusaSlicer
[image]

GuillaumeRives, 2020-08-11:
Ok, so it was a misunderstanding from me then. I used the standard Elegoo Mars profile and it works great !

closed 2020-08-11 by GuillaumeRives