sn4k3/UVtools #20: [BUG] Elegoo Mars profile does not work

darraghbr, 2020-08-12:
The Elegoo Mars profile does not appear to output correct parameters for the sliced file. 

The exported .sl1 file from PrusaSlicer looks fine when imported into UVtools, and the file looks fine when converted to .cbddlp and imported to UVtools, but when that file is imported into ChiTuBox the geometry appears to have been stretched. 

**To Reproduce**

1. Slice a model in PrusaSlicer using the UVtools Elegoo Mars profile
2. Export .sl1 file from PrusaSlicer
3. Import .sl1 file into UVtools
4. In UVtools convert the .sl1 file to .cbddlp
5. Import the .cbddlp file into ChiTuBox

**Expected behavior**

I expect the tool chain to provide a file that is representative and printable on the Elegoo Mars.

**Screenshots**

[image]

This is the model in PrusaSlicer. 

[image]

And this is the model when imported into ChiTuBox. As you can see the model has been stretched. 

**Files**

I have used the Ameralabs Town calibration part.

**Desktop (please complete the following information):**

Windows 10 build 19041

sn4k3, 2020-08-12:
Many people are afraid of this, but not really a problem, it's all good.
Please check this recent open issue: https://github.com/sn4k3/UVtools/issues/18

darraghbr, 2020-08-12:
Ok, that is a very strange bug...

darraghbr, 2020-08-12:
Re opened after editing issue to reflect it is just an appearance issue.

closed 2020-08-17 by sn4k3