sn4k3/UVtools #22: Use 4-connected-neighbor approach when evaluating islands

byancey, 2020-08-17:
**Is your feature request related to a problem? Please describe.**

In comparing result of islands identified by UVTools against islands identified by Photon Validator, I noticed that UVTools was missing islands that Photon Validator was finding.  On closer inspection, this appears to be related to components that are connected only on by a diagonal pixel, as below:

0000000000
0111100000
0111100000
0000011110
0000011110
0000000000

Photon Validator treats the above as two components, and evaluates each component's island status independently.  UVTools, on the other hand, treats this as one component, and evaluates it's island status based on that single component, such that if one side is well supported, the other side could be completely unsupported with effectively no bond to the supported portion of the component.

The solution for this is to use the 4-connected neighbors approach when detecting individual components rather than using the 8-connected neighbors approach.  That will ensure that components connected only by a diagonal will always be detected and treated as 2 (or more) separate components.

UVTools is currently using a combination of OpenCV's findContours and drawContours functions to identify individual components in a layer, and unfortunately, these functions will always uses 8-connected neighbors.  There is currently no way to configure them to use 4-connected neighbors.  However, OpenCV does have an alternative function called connectedContours which can take a parameter to specify 4 vs. 8 neighbor evaluation.  While the output of this function differs from findContour/drawContour, it can be used in a similar fashion to detect individual components within the layer.   

**Describe the solution you'd like**
I have already implemented a solution for this issue for your consideration to include in the master branch.  I will submit a push request for it once I have submitted this feature request.

My solution replaces calls to findContour and drawContour with a single call to connectedComponents, and also makes the necessary changes to the supporting logic to work with connectedComponents.  The general approach for island detection is largely unchanged from the current code.  I have also not modified any  code not related to island detection specifically, so detection of resin traps, for example, still relies on findContour and drawContour.

I have added a configuration item to the islandDetectionConfig which will allow the setting for 4-neighbor vs. 8-neighbor to be toggled.  This setting is not currently plumbed into the GUI, but changing the value in the code will allow island detection to be switched between the legacy 8-neighbor and new 4-neighbor modes (default is 4-neighbor).

I have tested has these changes in my local environment, and I have verified that when set to 8-neighbor mode, the island detection results are identical to  the current baseline behavior for UVTools.   This validates that these changes don't functionally impact the results of UVTools island detection.  In this mode, components connected by a single diagonal are treated as a single component, resulting in fewer islands being detected as compared to 4-neighbor mode. 

I have also verified that when set to 4-neighbor mode (default in my commit), the island detection results are functionally identical to Photon Validator.  UVTools actually reports more islands by count in this mode as compared to Photon Validator, but this is due to the fact that Photon Validator collapses all islands on the same layer into a single issue, whereas, UVTools adds a separate issue for each island, even when they share a layer.  I have walked through every island on a rather large model with a good number of islands and verified that UVTools and Photon Validator identify exactly the same islands within the model when this mode is enabled.

I propose that this change be incorporated in to UV Tools, either as the default, or as a configurable option in the GUI.

**Describe alternatives you've considered**

**Screenshots**

**Additional context**

sn4k3, 2020-08-17:
Nice find, thanks for detailed report, great work 👍 Will review the PR ASAP

sn4k3, 2020-08-17:
merged, will update GUI and release this patch

closed 2020-08-17 by sn4k3

byancey, 2020-08-17:
Thanks!Sent from my Verizon, Samsung Galaxy smartphone

sn4k3, 2020-08-17:
Findings: (6985 layers on 4K images) This approach took a little more to process islands (new 4C)107482 - (old 8C)89865 = 17617ms difference on a huge file with 66094 islands found (Much more islands detected now due no diagonals, most single pixels on bad AA diagonals which can be removed)
Comparing (new 8C) 95343 - (old 8C) 89865 = 5478ms more with 5990 islands found, much less islands then the 4C
This prove the method is working and the time loss compared with old method come with a gain and irelevant to wait this more

byancey, 2020-08-17:
Yes, neglected to mention that in my commit message, but that is consistent with my observations about performance as well. My first attempt, using only connectedComponents and iterating the full image was actually on order of magnitude slower (took a whopping 3 minutes vs. 15 seconds on my test model).  Your test model probably would have taken 15 minutes.  Fortunately I was able to identify some optimizations using connectedComponentsWithStats to get it to its current performance.This was my first time using OpenCV/Emgu, and I found the online documentation to be a bit lacking, particularly for the C# implementation.  It took some digging to determine that the "stats" returned by connectedComponents WithStats actually defines the bounding rectangle of each labeled component.  That allowed me to use an approach for components similar to how you approached contours, iterating only over the bounding rectangle of each component rather than the full layer. The end result was only slightly slower than the baseline, but as you said, a reasonable cost for the benefit. :)Thanks!--BryceSent from my Verizon, Samsung Galaxy smartphone

sn4k3, 2020-08-17:
CvInvoke repeats functions and arguments from OpenCV so documentation can be any of python opencv, problem is the connectedComponentsWithStats itself is lacking even arguments are kinda hidden on what they return. But anyway you did a good job for your first attempt with OpenCV. Fell free to contribute more or if you need any help on understanding the internals please contact.

byancey, 2020-08-17:
I'm consider a couple of other tweaks and will let you know if I move forward with anything to avoid duplication of effort.While I'm new to Open CV, and also C#, professionally I've been doing embedded software for about 25 years, so if I'm motivated enough,  usually I can manage to muck my way through something like this. :)  I actually prototyped my work for this in python, as it was much easier to dump arrays, etc. and make tweaks to familiarize myself with what the various OpenCV functions were actually doing before I delved into the C# implementation.I love it when open source projects overlap with my hobbies, as it allows me to make customizations for my own use,  even if they don't end up getting pulled back into master...and when they do,  all the better that the rest of the community can also benefit.I think what you've done with UVTools is pretty amazing.  At this point it has island detection parity with Photon Validator, with more advanced options for auto-removal, support for resin trap detection/removal, multi-file formats, etc.  I think there are some tweaks that could be made to the UI, etc. to make it a bit more intuitive and easier to use, but in my opinion, I think you've effectively made Photon Validator obsolete here. :oThanks again for the work you've done and for making this available to the community!

sn4k3, 2020-08-17:
> While I'm new to Open CV, and also C#, professionally I've been doing embedded software for about 25 years, so if I'm motivated enough,  usually I can manage to muck my way through something like this

That's amazing, almost my age on programing skills, and in fact that's reflected on the code and PR you've done.

> I actually prototyped my work for this in python, as it was much easier to dump arrays, etc.

Scripting is always easier to run with less, but if you use Visual Studio the debugging capabilities are awesome, you can set a breakpoint and debug every variable above, including arrays with the advantage of continue, step in/out etc, very powerfull tool and i use it all the time to check things out. 
C# is getting very strong and well built language with lot of eases today, very capable, the only downside with UVtools was i start with WinForm (Windows stuck), i never planed to build UVtools to this advanced stage, but someday i plan to convert it to Avalonia WPF which is cross platform GUI, the UVtools.Core and Cmd is already cross platform and ready to put on every OS.

> I love it when open source projects
Me too, open stuff make things go forward and better supported, almost all my works are open-source without any tradeoffs, make them for fun and help without the rage for money, so i'm giving all the work for free and support something that i don't take advange at the moment (I don't own a resin printer yet) but i actually have fun with this challenges.

> I think what you've done with UVTools is pretty amazing.  At this point it has island detection parity with Photon Validator, with more advanced options for auto-removal, support for resin trap detection/removal, multi-file formats, etc.

By your PR i guess you use and rely on uvtools for your prints? If so, can i ask how uvtools helped/fixed your files? What kind of usage you do with it (Workflow). I'm always looking for feedback on the software in order to improve...

> I think you've effectively made Photon Validator obsolete 
Don't blame me, Photon Validator obsolete itself by stop developing - Stop is dying ;)
UVtools objective was file convertion only, but then it evolved

> Thanks again for the work you've done and for making this available to the community!
My pleasure

byancey, 2020-08-18:
Yes, at this point I have used UVTools to post-process a few models before I printed them.

For resin prints, historically my typical workflow has been the following:

Prusa-Slicer (for supports) --> Chitubox for slicing --> Photon Validator for island processing --> Back to Prusa-Slicer and repeat the whole process if new supports are needed, otherwise print

Prusa-Slicer has always been superior for auto-supports, and Chitubox for manual supports, but you can’t generally use both in the same workflow without complications.  With proper support settings, Chitubox with manual supports can actually yield far superior results, with almost no model scarring, but it was much more time consuming, and Chitubox has been historically unstable.  It was actually my desire to slice in Prusa and eliminate Chitubox from my auto-support workflow that first led me to UVTools due to it’s ability to save cross-format, but I actually didn’t end up using it in that manner.  I installed it and played with it a bit initially, but decided not to use it in my workflow at that time.  In retrospect, it was new islands popping up due to the noise removal that kind of turned me off, but at the time I didn’t realize the root cause, and since I trusted Photon Validator, I stuck with it.

More recently, a new version of Chitubox was release with some very useful features, including auto support improvements, and the ability to visualize where supports are needed on the model.  With this newer version, I’ve found it much easier and more effective to add supports directly in Chitubox rather than relying on Prusa-Slicer for this step, so ironically I ended up eliminating Prusa-Slicer from my workflow instead of Chitubox, giving me the following workflow:

Chitubox --> Photon Validator --> Back to Chitubox if new supports are needed, otherwise, print.

What led me back to UVTools was actually a printer upgrade.  I replaced my 5.5” Epax X1 with a 9” Epax X10 4Kmono, only to discover that the new printer only supported ctb format.  This made it impossible to use Photon Validator for my island post-processing, and pretty much forced me to take a closer look at UVTools.  Once I figure out that the noise removal setting was removing the tips of some of my supports and introducing islands, and turned that setting off, I was able to post-process a few prints with UVTools and successfully print them…and, of course, further analysis of the islands found by Photon Validator when compared to UVTools led to the 4-connected-neighbor feature you just pulled into the baseline, so my current workflow is as follows:

Chitubox --> UV Tools --> Print

The nice part about this current workflow is that I’m able to use the more advance layer editing tools provided by UVTools to manually repair many of the issues not solved by auto-repair directly from UVTools.  This minimizes the number of times I have to go back to Chitubox, add a support, re-slice, and re-process islands.  Big time saver.  For example, in my last print, I was fairly quickly able to manually added a small support to the model that spanned about 10 layers.  Would have taken at least 10 minutes to go back to chitubox to add the support, reslice, etc., probably less than 30 seconds to add via UVTools.  I would never even have considered attempting such a thing in Photon Validator.

Hope that workflow explanation helps.

As far as other suggestions go, I have been playing around with the Layer Preview colors to find settings that I find easier to visualize having come from Photon Validator, and as part of that process, I’ve discovered that while there are more pixel types that can be colorized in UVTools, it’s currently not possible to replicate the colorization that Photon Validator provides.  

We have the option to see where new pixels will be added on the next layer (sloping out above – default cyan), where pixels were added on the previous layer (sloping out below – default magenta), and the case where a pixel was added below, and will also be added above (inward slopes meet – default red).  What we can’t currently visualize is new pixels that were added with the current layer that have no pixels below them (sloping in below).  Technically, you can see this information using the next layer color, but you are projecting a layer ahead in that case, so there are pixels being displayed on the current layer that haven’t technically printed yet, and when glancing at the layer, that can be a bit counter-intuitive.  With the ability to colorize new pixels on the current layer, you can get the same information without displaying pixels that haven’t printed yet.  I think simply adding another configurable color option for new pixels on the current layer that don’t have pixels beneath would give a great deal of flexibility in the layer preview, and though configuration, you could opt to see that info either on the current layer, or the next (or both, I suppose, though I wouldn’t likely configure it that way).  I guess for completeness we could consider colorization for pixels on the current layer that won’t have a pixel above (sloping in above), and the combination of both (outward slopes meet).  That said, where I think the first option would be useful, I’m less certain how useful those last two would be for layer analysis.

In any case, the above is something I’m considering looking into when I next have some free time, but if you it’s a quick fix for you, don’t wait on me.

[... cut ...] layer just shows the same islands in yellow, but moves the image around a bit.   Feels like it needs to be refined somehow…just not sure exactly how yet. 

Well, I didn’t expect this e-mail to be quite this long when I started my reply…but kept thinking of things to add as I went along.  In any case, hope that was helpful.  If you would like me to add issues to track some of these features requests to the UVTools repo in GitHub, I would be more than happy to add them.  Just let me know.

Thanks!

byancey, 2020-08-18:
After playing around with the color settings a bit more, I’ve amended my opinion on that topic.  Once I’ve trained my brain to register that the colors for next layer and previous layer are not actually part of the current layer, I think it’s probably a better way to visualize what’s going on from layer to layer than colorizing pixels on the current layer as I’m used to with Photon Validator…particularly if I configure the leading edge (next layer) as a light color closer to the white foreground, and the trailing edge (previous layer) as a dark color closer to the black background.   Works quite nicely when configured that way. I think it would still be a nice option to have the ability to configure the colors as I suggested below, but not certain I would even use it that way. 😊

sn4k3, 2020-08-18:
Thank you for sharing your experience, i see your workflow is now much simplified. I know some people who also stick to photon file validator, but after make the change to UVtools he don't turn back. If uvtool didn't exists a option to ctb people would be use other tool like uv3dp convert ctb to .photon correct on PFV and convert back to ctb.

About PrusaSlicer i guess next version will have the best support methods and place ahead any other slicer.

> Oh, the other suggestion I would have is to eliminate the progress bar at the bottom of the layer preview area.  

I wanted to get rid of it many releases ago but it always escape my eye because of it useless ^^ And yes i agree with you, it will be removed next version.

> The final thing I’ve thought about is tweaking the issues list to collapse multiple issues on the same layer into a single list, or doing something to make it easier to identify the specific island the issue is associated with when selected.  

List can be ordered and grouped, currently groups are limited to the type, but can be unlocked to layers too.
One way to solve the selected issue maybe use a darker color or other color?

About the colors preview on next and previous layer (Diference) i don't totally get what you are trying to achieve. But avaiable color selection is very open and can make use of 255*255*255 colors.

Fell free to open feature requests with your ideias and a topic per feature

byancey, 2020-08-18:
Reponses to a couple more of your questions below…

One way to solve the selected issue maybe use a darker color or other color?

I thought about that some more, and I think one simple answer would be to briefly blink or flash the pixels that compose the issue  when they are first selected.  I assume that’s probably fairly straight forward to do that with OpenCV, given that you effectively have the location and mask saved for each issue.  Currently when an issue is selected, the pixels in the issue are immediately zoomed and centered, so this would just add ‘blink-blink’ to the end of that process.  Zoom --> Center --> Blink-Blink.  That would probably be enough to draw the eye, and would avoid the need to add another distinct color to the layer preview.

About the colors preview on next and previous layer (Difference) i don't totally get what you are trying to achieve. 

Sorry, didn’t do a good job of explaining that.  For simplicity, let’s eliminate previous layer color from the equation and just focus on next layer color…that was really the only aspect of my previous attempted description that was relevant.  

 

Photon Validator visualizes new material growth that is not supported by the previous layer on the same layer on which that material was added.  UV Tools, on the other hand, visualizes new material growth by projecting what material is going to be added on the next layer.  Here’s a quick picture to demonstrate:

‘

 

In terms of a 3d print, what this means is that when you look at a layer in photon validator all the pixels on screen, regardless of color, will be cured on that layer.  In UVTools, on the other hand, there can be some colored pixels on screen that won’t actually be cured until the next layer.  Both approaches provide the same information, but visualize it a bit differently, so it really boils down to personal preference.  After working with these settings a bit more and refining my color settings, I actually think I prefer the UVTools approach.  That said, I believe adding one additional color setting to UVTools to set the color of pixels on the current layer that don’t have a pixel below them would effectively allow UVTools to be configured to replicate the Photon Validators approach.  Setting the “current layer growth” color to white would maintain the existing preview scheme.   Setting “current layer growth” to a color, while also setting the prev/next layer colors to black, would essentially replicate the photon validator approach (albeit with white rather than green as the main material color).

 

Not a real necessity, but it also doesn’t seem like it would be a difficult addition, so just boils down to whether you think it would be a useful addition.

 

Hope that explanation helped clarify.

 

Thanks!

byancey, 2020-08-19:
In my image,  layer 2 for photon validator, pixels should be yellow, not green.Sent from my Verizon, Samsung Galaxy smartphone

sn4k3, 2020-08-19:
Image is missing, but i think i got it now - Show next layer pixels that are unsupported by the current right?

byancey, 2020-08-19:
More precisely – Show current layer pixels that are unsupported by the previous.

 

I’ve included the image again, this time as an attachment rather than embedded.

sn4k3, 2020-08-19:
Still don't get the image... Use email to reply github sucks...

byancey, 2020-08-19:
Sent direct to your Hotmail address…

sn4k3, 2020-08-20:
I see it now, i think is all a matter of habituation, for me looking at PFV looks confusing, mostly because i'm used to UVtools.

byancey, 2020-08-20:
Yep.  Felt like would be a worthwhile addition when I first discovered the difference coming from Photon Validator...much less so now that I've gotten used to the UVTools approach.Sent from my Verizon, Samsung Galaxy smartphone