sn4k3/UVtools #25: Failed to run with Mono on Mac [BUG]

soswow, 2020-08-29:
**Describe the bug**
`./mono /Users/amotsjonov/Downloads/UVtools_v0.6.7.1/UVtools.exe` gives
```
WARNING: The Carbon driver has not been ported to 64bits, and very few parts of Windows.Forms will work properly, or at all
[ERROR] FATAL UNHANDLED EXCEPTION: System.TypeInitializationException: The type initializer for 'Emgu.CV.MatInvoke' threw an exception. ---> System.TypeInitializationException: The type initializer for 'Emgu.CV.CvInvoke' threw an exception. ---> System.DllNotFoundException: cvextern assembly:<unknown assembly> type:<unknown type> member:(null)
  at (wrapper managed-to-native) Emgu.CV.CvInvoke.RedirectError(Emgu.CV.CvInvoke/CvErrorCallback,intptr,intptr)
  at Emgu.CV.CvInvoke..cctor () [0x002fa] in <8bbe0921ca3f4d278fd57151609f0bb0>:0
   --- End of inner exception stack trace ---
  at Emgu.CV.MatInvoke..cctor () [0x00000] in <8bbe0921ca3f4d278fd57151609f0bb0>:0
   --- End of inner exception stack trace ---
  at Emgu.CV.Mat..ctor () [0x00000] in <8bbe0921ca3f4d278fd57151609f0bb0>:0
  at UVtools.GUI.FrmMain..ctor () [0x00000] in <acbc83e91056482ca0e570f4727ecef3>:0
  at (wrapper remoting-invoke-with-check) UVtools.GUI.FrmMain..ctor()
  at UVtools.GUI.Program.Main (System.String[] args) [0x00040] in <acbc83e91056482ca0e570f4727ecef3>:0
```

Running `./mono --arch=32 /Users/amotsjonov/Downloads/UVtools_v0.6.7.1/UVtools.exe` gives `Error: --arch=32 Failed to switch to './mono32'.`

Running `./mono32 /Users/amotsjonov/Downloads/UVtools_v0.6.7.1/UVtools.exe` gives `zsh: bad CPU type in executable: ./mono32`

**Desktop (please complete the following information):**
 - OS: MacOS 10.15.6
  
**Additional context**
I haven't restarted after installing Mono yet (can't do it right now. Will check after restart.

soswow, 2020-08-29:
Ok. It seems like 32 won't fly on Catalina, as they dropped support completely:

> macOS Catalina completely dropped support for 32-bit programs.

soswow, 2020-08-29:
https://github.com/mono/mono/issues/6701

sn4k3, 2020-08-29:
Won't work with UVtools

closed 2020-08-29 by sn4k3