sn4k3/UVtools #36: Prusa Profile Mars 2 Pro

SebSonic1979, 2020-09-05:
Hi, i just received my Mars 2 Pro and maybe i found an error in the Profile for Prusa Slicer. It seems that the value's for the X and Y Axis are twisted. When comparing the Values with the Values from Chitubox they are switched. Is this intended, can it be ignored or should it be changed to match the Settings from Chitubox?

sn4k3, 2020-09-05:
Ignore it, after open on UVtools check resolution, they just be right

SebSonic1979, 2020-09-06:
So, i did some test and my Mars 2 Pro shows the following behaviour:
STL sliced with Chitubox and then edited with UVTOOLS: gives me a Memory Error M 200000, does not start to print
STL sliced with Prusa Slicer and then edited with UVTOOLS: Print is starting

I just checked if the Printer starts to print, i will do a Test Print over Night to see if a Print Job will finish.

Hope you find the Problem why a Chitubox sliced File is failing, because in my oppinion Chitubox is better for doing manual supports than Prusa Slicer.

sn4k3, 2020-09-06:
Yes problem will be addressed next release

sn4k3, 2020-09-11:
Fixed

closed 2020-09-11 by sn4k3