sn4k3/UVtools #37: [FEATURE REQUEST] Layer Import

nvx, 2020-09-05:
**Is your feature request related to a problem? Please describe.**
As far as I know, the only way to author a photon file from scratch is to use the Photon File Editor tool which is pretty clunky.

**Describe the solution you'd like**
Being able to import new layers (from appropriately sized PNG images) into existing files, or create a new file from a series of images would be super handy to manually create pixel perfect slices.

sn4k3, 2020-09-05:
That was already on my todo list :)

nvx, 2020-09-05:
Awesome! What you've done with UVtools so far is amazing. As an exercise I supported and fixed up a really broken model exported from a game for a friend, definitely wouldn't have been able to fix it without UVtools (or remaking the model entirely). I never realised there was anything missing from the Photon File Validator until I checked out UVtools 😅

sn4k3, 2020-09-05:
You can only know what you are missing if you try different things ;)

sn4k3, 2020-09-07:
Preview of the tool, if you have more ideas to add just write

[image]

nvx, 2020-09-07:
Awesome. I assume that it'd support AA by importing files with shades of grey not just black and white pixels yeah?

Also I assume the add button lets you select multiple files at once not just one at a time?

sn4k3, 2020-09-07:
Yes and Yes :)

sn4k3, 2020-09-11:
Implemented

closed 2020-09-11 by sn4k3