sn4k3/UVtools #38: [FEATURE REQUEST] 3D View from slice

nvx, 2020-09-05:
**Is your feature request related to a problem? Please describe.**
When fixing problems with a file, sometimes it's handy to be able to visualise the layers in 3 dimensions.

**Describe the solution you'd like**
Ability to view the file in 3D as rendered voxels up to a certain layer.

**Describe alternatives you've considered**
The Photon File Validator tool has functionality to do this, but is pretty clunky in execution. Slicers often allow this visualisation too (although they use the original STL not the sliced version for visualisation - which wouldn't be much use when trying to see the effects of pixel edits).

sn4k3, 2020-09-05:
I had plans for this, but WinForms are bad for 3D rendering :( I still will look for alternatives

eried, 2020-09-23:
It would be awesome to see the issues there :P in 3d like a radiography

eried, 2021-01-13:
BTW, as an alternative Chitubox creates a 3d view of the sliced file (even after processing it with UV tools)

dgm3333, 2021-04-18:
Because the chibubox format basically lists the start and end pixels in each row of each layer it would be pretty easy to use this list to create short lines then just display them live using a pupose built 3D tool such as blender or FreeCAD.
Blender is already easily handles displaying 1000s of layers of a slice and I am already using it to display and repair mesh errors and layers which have errors following slicing the mesh directly with Blender (see image below).<p>
sn4k3 advised that UVTools exposes it's functions as a dll (https://github.com/ezrec/uv3dp/issues/142) but I haven't got this working yet from Blender although I not sure where the issue is (see this post https://stackoverflow.com/questions/67122384/no-module-named-error-when-attempting-to-importing-c-sharp-dll-using-python-ne), but when it is working it's likely to be very quick to implement.<p>
Using blender or FreeCAD would also mean it would be possible to fix the source mesh directly as it can also linked with tools such as MeshLab (eg see this post: https://blenderartists.org/t/meshlab-accessible-direct-from-blender-python/1299022/2).
slic3r, PrusaSlicer, Chitubox, etc. Theoretically PrusaSlicer does repair and export meshes (and has a great per slice 3d view cf chitubox), but even after slicing it still leaves unprintable errors - eg the two views are only 1 layer different, and occur due to non-manifold meshes and single layer faces (ie trying to effectively slice something with zero layer thickness) AFTER an attempted automated repair.<p>
There's unlikly to be a "one size fits all" solution, and at the moment many tools work reasonably well although very few slicers are really optomising printing with resin printers since they were mostly founded in FDM, but Chitubox (which AFAIK wasn't) is one of the more basic tools. Since they're all free and opensource I'd personally like something which integrated the UVTools repair strategies into PrusaSlicer which could then could be accessed as a library by Blender or FreeCAD to show mesh issues and support repair prior to reslicing and printing...

[image]
[image]

sn4k3, 2021-04-18:
> Because the chibubox format basically lists the start and end pixels in each row of each layer it would be pretty easy to use this list to create short lines then just display them live using a [...]

It really doesn't matter how the format is made, if you have the raw image then you can do any kind of transformation and faster than have to read RLE and decode it into something. Render single lines or pixels are slower, you can just render all the image and stack it. UVtools even work with contours which are like vectors, image pixels turn into connected points.
Right now the UI framework have a bug on OpenGL that prevent me from using it. So i can't have custom 3D renders atm.

> sn4k3 advised that UVTools exposes it's functions as a dll

`"UVtools is all built as a library, but **require a C# project**, you can import the UVtools.Core.dll"`
That means UVtools.Core is a C# Library, which can be loaded but from a .NET project/environment. For example, to use it you must create a dotnet console app and import the dll. 
About python i think it will not work, because python.net don't mention NET5.0. And on my python3.9 clr doesn't work
The only way to make it work outside .NET world is to expouse classes and methods with COMVisible (I never tried that, but i think that is the way)
If you can use any language to do the work then why not try the C# approach? Less flexibility than python for sure, but still very easy to deal.

> Using blender or FreeCAD would also mean it would be possible to fix the source mesh directly as it can also linked with tools such as MeshLab (eg see this post: [...]

PrusaSlicer uses netfabb to do the repair, but most of the problems come from the model design.
UVtools doesn't accept .stl format and even if implement a stl repair function i think it not worth to have it separated from slicer, so the workslow would be: UVtools -> repair stl -> import stl into slicer -> slice ...
What is desired is: Slicer -> import stl (auto-repair or manual) -> slice ... (so you cut some steps)
With prusa slicer you can repair a model by right click on it, when errors are detect it will show a warning sign near the model file

> There's unlikly to be a "one size fits all" solution, and at the moment many tools work reasonably well although very few slicers are really optomising printing with resin printers since they were [...]

The only opensource slicer for msla is the PrusaSlicer.
Chitubox is lame
Then you have Lychee which is all made for msla
And a new slicer is comming with a amazing layout, his dev is very receptive to features and ideias, it may worth to contact him: https://www.facebook.com/groups/uvtools/permalink/452910025818506/

MartinLunn, 2021-04-23:
I would appreciate this feature as well.

I've tried using blender and chitubox as a reference for the 3d object (mostly for island fixing).

Comparing the 2d slice from UV tools and the 3d image from blender I find I don't always know where  the 2d slice is in the 3d space. I can't tell what height it's at with fine enough resolution.

Using Chitubox I get a better sense because I can enter the layer number, Chitubox does a horrible job of isolating single layers. In it's current iteration, it has two draggable sliders to select the bottom and the top of the visible slice:

[image]

This is fine, but when you want to select only a few layers by entering the layer numbers, it's almost impossible to select which slider you want, so you end up fighting the UI.

Obviously it would be nicer if one could move the view around the 3d preview, but even a static preview would be helpful.

What I REALLY want is the 3d preview with the current 2d layer highlighted, and the above and below layers highlighted in different colours.

sn4k3, 2021-04-23:
Right now i can't have 3D renders on the app, the UI framework have a bug that prevent me from using OpenGL viewer. 
While is possible to emulate a 3D view with 2D it will be extremly slow to apply live modifications and forget about orbit!
All hope now is that they will fix it or introduce a render for angle

MartinLunn, 2021-04-23:
Fair enough!

I would be totally fine with slow live modifications. Like if I went into the pixel editor, made some changes, hit apply, and came back 30s later, that's fine by me.

As an aside, I found sometimes the 2d layer view in UV tools would be mirrored as compared to the slicer. I'm pretty sure it's a user error, but maybe a mirror button would be useful.

I'll give prusaslicer a shot!

sn4k3, 2021-04-23:
The mirror on UVtools is called Flip
[image]

tslater2006, 2021-12-18:
For what it's worth, UVTools now has an export to mesh feature and then you can use any mesh viewing application to view it. It's not the full request but at least half of it

sn4k3, 2026-09-19:
...only 6 years later, one of the very first feature requests is finally implemented! 🥳

Not bad... could've taken longer 🤣

Discussion here: https://github.com/sn4k3/UVtools/discussions/1148

closed 2026-09-19 by sn4k3