sn4k3/UVtools #50: [BUG] Crash on launch

ComteNiark, 2020-09-12:
The application does not launch, generates a crash dump. OS is updated windows 10. I also tried with 8.0 since I saw the error was referencing a path that wasn't on my computer so I thought the path was hardcoded by dev mistake, but since both latest versions had the same error, I'm not so sure anymore.
I have tried to update NET framework, but I already had a more recent version.

Crash dump:

```
[UVtools]
Date=12/09/2020 05:58:08
Application=UVtools 0.8.1.0
ExecutablePath=D:\Program Files (x86)\UVTools\UVtools.exe
Company=PTRTECH

########################
##        Error       ##
########################
[Exception1]
Message=The type initializer for 'Emgu.CV.MatInvoke' threw an exception.
StackTrace=   at Emgu.CV.MatInvoke.cveMatCreate()
   at UVtools.GUI.FrmMain..ctor() in D:\Tiago\Dropbox\Programming\C#\UVtools\UVtools.GUI\FrmMain.cs:line 97
   at UVtools.GUI.Program.Main(String[] args) in D:\Tiago\Dropbox\Programming\C#\UVtools\UVtools.GUI\Program.cs:line 90
TargetSite=IntPtr cveMatCreate()
Source=Emgu.CV.Platform.NetStandard
HelpLink=

[Exception2]
Message=The type initializer for 'Emgu.CV.CvInvoke' threw an exception.
StackTrace=   at Emgu.CV.CvInvoke.CheckLibraryLoaded()
   at Emgu.CV.MatInvoke..cctor()
TargetSite=Boolean CheckLibraryLoaded()
Source=Emgu.CV.Platform.NetStandard
HelpLink=

[Exception3]
Message=Unable to load DLL 'cvextern': The specified module could not be found. (Exception from HRESULT: 0x8007007E)
StackTrace=   at Emgu.CV.CvInvoke.RedirectError(CvErrorCallback errorHandler, IntPtr userdata, IntPtr prevUserdata)
   at Emgu.CV.CvInvoke..cctor()
TargetSite=IntPtr RedirectError(CvErrorCallback, IntPtr, IntPtr)
Source=Emgu.CV.Platform.NetStandard
HelpLink=

########################
## System Information ##
########################
[System]
OperativeSystem=Windows 8 x64
ProcessorCount=12

########################
##     Open Forms     ##
########################
[Forms]

########################
## Loaded Assemblies  ##
########################
[mscorlib, Version=4.0.0.0, Culture=neutral, PublicKeyToken=b77a5c561934e089]
   Location=C:\Windows\Microsoft.NET\Framework64\v4.0.30319\mscorlib.dll

[UVtools, Version=0.8.1.0, Culture=neutral, PublicKeyToken=null]
   Location=D:\Program Files (x86)\UVTools\UVtools.exe

[System.Windows.Forms, Version=4.0.0.0, Culture=neutral, PublicKeyToken=b77a5c561934e089]
   Location=C:\Windows\Microsoft.Net\assembly\GAC_MSIL\System.Windows.Forms\v4.0_4.0.0.0__b77a5c561934e089\System.Windows.Forms.dll

[System, Version=4.0.0.0, Culture=neutral, PublicKeyToken=b77a5c561934e089]
   Location=C:\Windows\Microsoft.Net\assembly\GAC_MSIL\System\v4.0_4.0.0.0__b77a5c561934e089\System.dll

[System.Drawing, Version=4.0.0.0, Culture=neutral, PublicKeyToken=b03f5f7f11d50a3a]
   Location=C:\Windows\Microsoft.Net\assembly\GAC_MSIL\System.Drawing\v4.0_4.0.0.0__b03f5f7f11d50a3a\System.Drawing.dll

[ApplicationManagement, Version=1.0.0.0, Culture=neutral, PublicKeyToken=null]
   Location=D:\Program Files (x86)\UVTools\ApplicationManagement.dll

[UVtools.Core, Version=0.8.1.0, Culture=neutral, PublicKeyToken=null]
   Location=D:\Program Files (x86)\UVTools\UVtools.Core.dll

[netstandard, Version=2.0.0.0, Culture=neutral, PublicKeyToken=cc7b13ffcd2ddd51]
   Location=C:\Windows\Microsoft.Net\assembly\GAC_MSIL\netstandard\v4.0_2.0.0.0__cc7b13ffcd2ddd51\netstandard.dll

[Emgu.CV.Platform.NetStandard, Version=4.4.0.4061, Culture=neutral, PublicKeyToken=7281126722ab4438]
   Location=D:\Program Files (x86)\UVTools\Emgu.CV.Platform.NetStandard.dll

[Cyotek.Windows.Forms.ImageBox, Version=1.2.0.0, Culture=neutral, PublicKeyToken=58daa28b0b2de221]
   Location=D:\Program Files (x86)\UVTools\Cyotek.Windows.Forms.ImageBox.dll

[ObjectListView, Version=2.9.1.25410, Culture=neutral, PublicKeyToken=b1c5bf581481bcd4]
   Location=D:\Program Files (x86)\UVTools\ObjectListView.dll

[System.Runtime.InteropServices.RuntimeInformation, Version=4.0.0.0, Culture=neutral, PublicKeyToken=b03f5f7f11d50a3a]
   Location=C:\Windows\Microsoft.Net\assembly\GAC_MSIL\System.Runtime.InteropServices.RuntimeInformation\v4.0_4.0.0.0__b03f5f7f11d50a3a\System.Runtime.InteropServices.RuntimeInformation.dll

[System.Configuration, Version=4.0.0.0, Culture=neutral, PublicKeyToken=b03f5f7f11d50a3a]
   Location=C:\Windows\Microsoft.Net\assembly\GAC_MSIL\System.Configuration\v4.0_4.0.0.0__b03f5f7f11d50a3a\System.Configuration.dll

[System.Core, Version=4.0.0.0, Culture=neutral, PublicKeyToken=b77a5c561934e089]
   Location=C:\Windows\Microsoft.Net\assembly\GAC_MSIL\System.Core\v4.0_4.0.0.0__b77a5c561934e089\System.Core.dll

[System.Xml, Version=4.0.0.0, Culture=neutral, PublicKeyToken=b77a5c561934e089]
   Location=C:\Windows\Microsoft.Net\assembly\GAC_MSIL\System.Xml\v4.0_4.0.0.0__b77a5c561934e089\System.Xml.dll
```

sn4k3, 2020-09-12:
I reisntalled via MSI on my pc and works file.
Crash is telling cvextern is not loaded, can you confirm if you have that file under x64 folder under program folter?
Maybe take a screenshot of that folder content and post here.
Also have you tried run as administrator?

ComteNiark, 2020-09-12:
I installed via zip because I wanted to install in my secondary hard drive (C: is a small sized SSD I use for fast boot/load times)

Here is a screenshot of the folder's contents: 
https://prnt.sc/ufx6ke
As you can see, cvextern.dll is present. I literally just dumped the zip in my D:/Program Files (x86) folder.

ComteNiark, 2020-09-12:
Running as administrator gives this error log:

`########################
##        Error       ##
########################
[Exception1]
Message=The type initializer for 'Emgu.CV.MatInvoke' threw an exception.
StackTrace=   at Emgu.CV.MatInvoke.cveMatCreate()
   at UVtools.GUI.FrmMain..ctor() in D:\Tiago\Dropbox\Programming\C#\UVtools\UVtools.GUI\FrmMain.cs:line 97
   at UVtools.GUI.Program.Main(String[] args) in D:\Tiago\Dropbox\Programming\C#\UVtools\UVtools.GUI\Program.cs:line 90
TargetSite=IntPtr cveMatCreate()
Source=Emgu.CV.Platform.NetStandard
HelpLink=

[Exception2]
Message=The type initializer for 'Emgu.CV.CvInvoke' threw an exception.
StackTrace=   at Emgu.CV.CvInvoke.CheckLibraryLoaded()
   at Emgu.CV.MatInvoke..cctor()
TargetSite=Boolean CheckLibraryLoaded()
Source=Emgu.CV.Platform.NetStandard
HelpLink=

[Exception3]
Message=Unable to load DLL 'cvextern': The specified module could not be found. (Exception from HRESULT: 0x8007007E)
StackTrace=   at Emgu.CV.CvInvoke.RedirectError(CvErrorCallback errorHandler, IntPtr userdata, IntPtr prevUserdata)
   at Emgu.CV.CvInvoke..cctor()
TargetSite=IntPtr RedirectError(CvErrorCallback, IntPtr, IntPtr)
Source=Emgu.CV.Platform.NetStandard
HelpLink=
`

Same thing as far as I can tell.

sn4k3, 2020-09-12:
Very odd situation, i tried to replicate your problem but here runs fine.
Try another test: unzip to D:\uvtools and run from there

ComteNiark, 2020-09-12:
The Program Files (x86) folder on D: is not a special folder, I just named it like the original programs folder in C: so most installers would find it by changing only one letter in the install, because I'm lazy like that.

I did just attempt what you said though, because you never know and sure enough, same error. I'll try on C:, give me a minute.

ComteNiark, 2020-09-12:
Rebooted and re-tried with the UVTools folder at the root of the C: drive, still throws the same error.
Is the cvextern from some redistribuable? Maybe I could download it from a primary source to see if that fixes it?

sn4k3, 2020-09-12:
Ok, try to move all x64 contents to uvtools root. Re run

ComteNiark, 2020-09-12:
I had already suspected that and tried it. I did it again because now I'm running it on C: so maybe that changed something and... nope. Still the same error code. It seems it just can't hook into this specific dll for some reason.

ComteNiark, 2020-09-12:
Same thing with x86 dlls, but that was to be expect since I'm running 64 bit windows anyway.

sn4k3, 2020-09-12:
It looks like it cant find the dlls for some reason.
Can you try to copy them to System32 folder?

ComteNiark, 2020-09-12:
Out of curiosity I tried it with an old release (6.5 I believe), still the same results...
Will do, reporting back in a bit.

sn4k3, 2020-09-12:
How old is your windows install and do you perform any modification?

ComteNiark, 2020-09-12:
The dll are already present in System32, same size and everything. I can't actually replace them with your version though because System32 shows up as "in use" and won't allow me to copy files over to it.

My windows install is fresh from about a month ago. Installed off the official windows iso. I have done a few modifications but all are cosmetic in nature (start menu, different file explorer GUI) and none have ever stopped a piece of software from running until now.

sn4k3, 2020-09-12:
Wait, you had those dlls already present on system32? cvextern and so?.
If so you have to delete them

ComteNiark, 2020-09-12:
Yes, those were already present in System32. Googling around, I found those are apparently openCV dll files? I am doing an install of the openCV runtime right now to see if that fixes the problem. If not, I will try to find a way to replace those DLL files with yours (get around the "folder in use" issue). Will report back in a bit.

sn4k3, 2020-09-12:
Thats the reason why it cant load cvextern, it's being used by some other app...
uvtools never copy dlls to system32 so it must be other app in conflict.
https://lockhunter.com/ with that tool you can see what is locking the files, please run that to discover what app is consuming that dlls

ComteNiark, 2020-09-12:
Thanks for the tool, that will come in handy for more than this bugfix! Let me give it a go.

ComteNiark, 2020-09-12:
Replaced all the DLLs in System32 (msvcp_140.dll was the one being locked up by a node.js install from adobe...), still no luck, still the same error. I will try to reboot my computer to check, but as you can see from the modified date, these are both the same file: 
[image]

sn4k3, 2020-09-12:
Remove them from system32 instead of replace.
Also i would find the whole drive for that cvextern.dll

ComteNiark, 2020-09-12:
I have deleted the adobe program using one of the DLLs. I have also deleted every DLL from system32, individually.
Searching for cvextern.dll on my entire computer (2 drives) now only returns results from my installs of UVtools, one of which (D:) I have now deleted, just to be sure.
[image]

 Aaaand... still the same. I think I'm doomed not to use this piece of software...

sn4k3, 2020-09-12:
this is the first case i saw with this problem.
I need to go sleep now, but tomorrow i can build special version for you try, in order to debug this.
My last advice for today: Go to cvextern.dll properties and see if you have file permitions under security, odd but we never know

ComteNiark, 2020-09-12:
I'll try some stuff on my side too, hope we can solve this because I am currently writing a guide for the /r/anycubicphoton subreddit and would like to feature the program if it works better than what the community currently uses, which looking at your feature list seems to be the case.

As you can see, administrators have full rights: 
[image]

sn4k3, 2020-09-12:
As you are run as user you must check Users. Administrators are for Run as Administrator

ComteNiark, 2020-09-12:
I am launching the .exe as administrator though. I will still try this however.

ComteNiark, 2020-09-12:
Oooh, I think I found a clue! For some reason, the UVTools folder is marked as read only, and I cannot change it. I tried a clean install from a freshly downloaded zip and it's still the same. Unchecking "read only" from the folder properties does not work, as soon as I close and open properties again it it's checked again.

sn4k3, 2020-09-12:
Copy program to a pen and try on other machine, or VM if you have any. If work out of the box there i'm sure is a problem inside your system

sn4k3, 2020-09-12:
Read-only with square check is OK, uvtools don't need write permitions but in fact you have them or else you wont write logs to AppLogs ^^

ComteNiark, 2020-09-12:
Ah, guess you're right. I tried a clean install from the msi but yes, it's still read only, and yes, the issue is still present. I'll install a virtual machine to see if that changes anything.

ComteNiark, 2020-09-12:
The plot thickens... I tried using something called Dependency Walker to check cvextern, see if I was missing a runtime needed for it and... Well..

[image]
So something may be wrong with my system, maybe? Idk what though. I'll keep digging.

ComteNiark, 2020-09-12:
Turns out that was a false positive because Dependency Walker is an old program... Have to keep searching.

ComteNiark, 2020-09-12:
According to [this ](https://en.it1352.com/article/40079206a3f248eca76213aac0de15f9.html), you have to include all dependencies and if you don't, it will show up as a cvexternal.dll issue... I assume this is some dependency most other people have by default but I don't have for some reason because I probably deleted it while cutting some windows bloatware... Could you make a release with all the opencv/EMGU dlls included?

ComteNiark, 2020-09-12:
I downloaded EMGU and put every dll included in the release in UVTools' root folder and still the same error. I think I'll have to give up... Too bad.

sn4k3, 2020-09-12:
uvtools auto include all required dependencies, no more no less.
Have you tried on a VM?
Tried with antivirus off?

ComteNiark, 2020-09-12:
I did not run it in a VM because virtualization is disabled in BIOS and I honestly couldn't be asked to go to BIOS and turn it on. What I did was try it on a laptop I had laying around, and it worked, so the problem is definitely with my machine, though I fail to understand what it is.

I have no antivirus to speak of, even windows defender is disabled. I only run malwarebytes occasionally if I ever suspect something is wrong, which I did after this whole affair and no results.

sn4k3, 2020-09-12:
Well it must be something,  my system is old and with tons of patchs from updates and versions, have tons of software installed and services, adobe included and still never had this problem. If you have nothing to lose maybe reformat your pc is the fastter way

ComteNiark, 2020-09-12:
I believe that is the only option left... I'll have to redo my extensive windows modding though... I'll try it and report back tomorrow I think.

ComteNiark, 2020-09-12:
FIXED. Christ.
The problem was that I'm running on an "N" installation of windows 10, which cuts out some bloatware media programs. CVextern.dll has 3 of the DLLs removed by the N edition as one of its dependencies. Installing the media pack from windows update fixed the issue after a reboot.

closed 2020-09-12 by ComteNiark

sn4k3, 2020-09-12:
I known emgucv requires the media pack, but wouldn't expect someone to have N version of windows, also didn't know the effect by not having the media pack. Noted now, will add it as requirement on readme.

ComteNiark, 2020-09-12:
I try to keep my install as clean as possible, so cutting out some default-install software I did not need (since I use discord and MPC-HC) was seen as a convenient option... Had I known better I would've picked a normal install haha!

sn4k3, 2020-09-12:
N version only exists because some stupid rules to not force software to users to have the "freedom" to opt but them windows come bloated with Apps who paied to be there... The problem is some libraries will go away too.

ComteNiark, 2020-09-12:
Ah, well I guess I am one of those users who need the freedom 😄. Anyway, this issue is good for now. I'll review the software and write the guide as planned, from what I'm looking at it does look better than what our community is currently using. With a bit of luck you'll get a lot of new users ;)
Good luck on development.

sn4k3, 2020-09-12:
Freedom always come at a cost ;)
You were using Photon File Validator i guess?

ComteNiark, 2020-09-13:
Yes, those of us bothering to validate our files were using Photon File Validator, but since it's not being updated anymore and it seems to have glitches in the latest version regarding antialiasing (which means it can only be used for the original Photon, not Photon S or Zero), it'll be glad to aim the community towards an active project :)