sn4k3/UVtools #59: [BUG] Longer Orange 30 and Lychee Slicer

calipzo, 2020-09-13:
**Describe the bug**
When I open a file silced in Lychee Slicer it looks incorrect.

**To Reproduce**
Steps to reproduce the behavior:
1. Install Lychee Slicer (free version; v3.0.11 at this time).
2. Add Longer Orange 30 printer.
3. Slice proposed model with default settings.
4. Open it in UV Tools.

(my file in the bottom)

**Screenshots**
[image]

**Files**
[Here is my *.lgs30 file](https://mega.nz/file/bE0yEQ6Y#0iLsrYnghKpptmU1XB4L3HXvhiZ2zvYvu4tLH2znohI) sliced with Lychee Slicer (if needed).

sn4k3, 2020-09-13:
Resolution is fliped (bad configuration on slicer, invert resolutions).

calipzo, 2020-09-13:
It's very strange. I have [found on my flash old sliced file](https://mega.nz/file/kk5RGJZQ#2hgK-edaw9OGWPRtEZlyNUKGmHeQXr9h_PD4UEUoDdw) (sliced with Chitubox) - and it have the same problems. The file has been printed many times without problems.

[image]

sn4k3, 2020-09-13:
Can you show your chitubox printer configuration dialog?

sn4k3, 2020-09-13:
This is what i got from pre-configurated machines on chitubox. As you can see your X is fliped with Y
UVtools shows: 2560x1440 and the correct is the flip value
[image]

sn4k3, 2020-09-13:
Can you send me a old file sliced with chitubox?

calipzo, 2020-09-13:
Old file: [download here](https://mega.nz/file/kk5RGJZQ#2hgK-edaw9OGWPRtEZlyNUKGmHeQXr9h_PD4UEUoDdw). I have sliced it 4+ months ago. At the time, both the Chitubox and the plugin were older versions.

In Chitubox I use standard settings except exposure time and layer height.
[image]

sn4k3, 2020-09-13:
Yes that file is flipped too, the implementation was based on new chitubox version + plugin. Can you try that on your machine just to confirm?

calipzo, 2020-09-13:
Ok, i will report.

vknvkz, 2020-09-18:
Hi. Same problem with ChituBox 1.6.5.1 and Longer3D plugin. Default Chutu profile for Orange30 printer was used

[image]
[_20x20x20.lgs30.zip](https://github.com/sn4k3/UVtools/files/5243222/_20x20x20.lgs30.zip)

sn4k3, 2020-09-18:
Thats so strange, can you point me out the URL of plugin you are using for LGS?

sn4k3, 2020-09-18:
In meanwhile here's a compiled version with XY instead of YX: https://easyupload.io/k0q599
Use that while i try to discover why my chitubox+plugin gave me good results with YX, which never made sense in first place

vknvkz, 2020-09-18:
> Thats so strange, can you point me out the URL of plugin you are using for LGS?

I was using V1.2 plugin come on usb flash with printer.
[image]

I found newer V1.4 plugin: https://drive.google.com/drive/folders/1uMnSBDKeK9if6TdWSsscwQ9E6BxCG6rK?fbclid=IwAR0gWYL_AOq1SRZiQqV2XjSR14KmszEzrn_3I0Z9Hh_qnWu17yBRTlNC5iU

It work fine with original 0.8.2.2 version.

(V1.2 plugin must be deleted first, do not import V1.4 over V1.2!)

sn4k3, 2020-09-18:
What a mess they did... I'm gona fix this by put a auto resolution flip on code, it will work for both cases

sn4k3, 2020-09-19:
Implemented

closed 2020-09-19 by sn4k3