sn4k3/UVtools #60: Can't convert a file

ComteNiark, 2020-09-14:
Back with another bug that probably only affects me...

Whenever I try to convert a SL1 file to another format (mainly .photon but I've also tried .pws), I get this error; 
[image]

I assume photon conversion is implemented so that probably isnt the issue... What I don't get is the second part of the error message. I can't even begin to understand what that means.

I've tried running as admin, didn't change anything.
What I'm doing in the meantime is saving as SL1 and then converting the the SL1 to Photon converter, but that's not optimal when UVtools could just do it for me.
Thank you for your help.

sn4k3, 2020-09-14:
Always attach the input file when bug is related to them.
Most likely you are using Prusa SL1 printer, you MUST use UVtools printers.
Go to UVtools -> Help -> Install profiles

ComteNiark, 2020-09-14:
Wow, you're right. I assumed I didn't need the new profiles because I already had a Photon sized printer set up in my prusaslicer and that they were just something nice to get you started using prusa. I was wrong. Sorry for the trouble.

closed 2020-09-14 by ComteNiark

sn4k3, 2020-09-14:
Prusaslicer lacks of extra settings need for chitubox formats, such as lift speed etc, so i must come with a solution, those values are set user Printer -> Notes

ComteNiark, 2020-09-14:
That's a very ingenious method! I'll be sure to edit the notes if I ever need to change these settings. Maybe you should change the error message that I encountered with an explanation of this? Because even a savvy user like me had no idea what to do.

I hope future versions of prusaslicer will allow us to slice directly to other formats, but they seem a bit lost on what the standards are since SLA is all new for them, and development has been sluggish due to Covid...

sn4k3, 2020-09-14:
Yes i think i can improve that error message.
I'm sure prusaslicer will do that