sn4k3/UVtools #62: [BUG] New W/S hotkeys eat keystrokes inside text input field

tofof, 2020-09-18:
**Describe the bug**
The 'text' pixel editor can no longer receive the W or S keystrokes when typing in the text field.

**To Reproduce**
Steps to reproduce the behavior:
1. Pixel editor -> Text -> Text: 
2. Type 'qwer asdf'
3. See that you've only typed 'qer adf'

Home/end work as expected for text field, but also jump to top/bottom layer at same time.

**Expected behavior**
All (?) hotkeys disabled while typing in text field

**Desktop (please complete the following information):**
 - OS: Win 10 Pro x64 1803
 - UVTools 0.8.2.2

Keep up the great work! I was thrilled by the region-of-interest addition.

sn4k3, 2020-09-18:
Confirmed, will be addressed on next patch

sn4k3, 2020-09-19:
Fixed

closed 2020-09-19 by sn4k3