sn4k3/UVtools #63: [FEATURE REQUEST] Erase drawing edits

tofof, 2020-09-18:
**Is your feature request related to a problem? Please describe.**
I click to draw with e.g. an 8 pixel brush and misjudge exactly where I need to click. 
If I notice immediately I can fix it by removing edits from the top of the list until it's fixed -- however, this wasn't intuitive.
But if I notice after doing several more edits it's very difficult to determine which ones would fix it.

**Describe the solution you'd like**
Erase edit mode - draw with cursor to return green or red pixels to the original unedited state

**Describe alternatives you've considered**
Cursor that shows bounds of area you're going to draw in would also help:
[image]

**Additional context**
The text for brush size should probably be "diameter ... px" instead of "area ... px<sup>2</sup>", since setting it to 5 gives you a 5x5 cursor of 25 px<sup>2</sup> area.

sn4k3, 2020-09-18:
You mean after release button pixels are shifted right?

About the labels i didnt understand what's wrong, if you set 5px<sup>2</sup> that's equal to 25px diameter so it's correct labeled.
Or do you mean turn into pixel diameter instead to be more intuitive? If so theres a big problem with that, a square is given by x * y, lets supose user enter 3px, that will turn into 1.5 * 1.5, and theres no fraction of pixels and that will be cast to 2*2 misplacing and invalidate user input. So diameter must be even number for both square and circle. That's why i give area instead

tofof, 2020-09-18:
> You mean after release button pixels are shifted right?

No. On one of the 0.7 versions that would happen, but that's not what I'm talking about (and already seems to be fixed.)
I just mean my own error - if click in the wrong place, it would be nicer to fix that by being able to click and drag with a modifier key to erase the green/red pixels, instead of trying to find the right edits in the list to choose 'remove' on.
The better cursor would eliminate some of my error by letting me see better what will be painted in, but doesn't help if I just click badly.

Regarding the label:
Your example is correct - asking user for area just creates problems. 
We both agree the user should input diameter.
But right now, the text says "area"/"px<sup>2</sup>". 
It should say "diameter"/"px" since that's what you actually want user to input.

>if you set 5px2 that's equal to 25px diameter so it's correct labeled.

That sentence is wrong, which is the problem. You have the words swapped.
5 = diameter.  
25 = area.

tofof, 2020-09-18:
Maybe part of the confusion is the "px<sup>2</sup>" part?

5 meters * 5 meters = 25 square meters = 25 m<sup>2</sup>
I interpreted px<sup>2</sup> like this.

But maybe you were trying to indicate something like
[   5 m   ]<sup>2</sup>   =  25 square meters

It's extra weird because pixels are actually already a 2-dimensional unit, unlike meters, so px<sup>2</sup> is a strange thing to write anyway.

sn4k3, 2020-09-18:
`No. On one of the 0.7 versions that would happen, but that's not what I'm talking about (and already seems to be fixed.)
I just mean my own error - if click in the wrong place, it would be nicer to fix that by being able to click and drag with a modifier key to erase the green/red pixels, instead of trying to find the right edits in the list to choose 'remove' on.
The better cursor would eliminate some of my error by letting me see better what will be painted in, but doesn't help if I just click badly.`

I see now, like a rubber? To erase/undo your drawings and keep the original behind?

Yes you are right about the labels it will fix that

tofof, 2020-09-18:
> I see now, like a rubber? To erase/undo your drawings and keep the original behind?

Exactly.

sn4k3, 2020-09-18:
Ok, about the cursor preview, it will be implemented on the next generation of UVtools (v1), right now no viable option without impacting performance and no alpha to do it right

sn4k3, 2020-09-19:
implemented

closed 2020-09-19 by sn4k3