sn4k3/UVtools #64: [FEATURE REQUEST] Possible overhang detection

eried, 2020-09-23:
Awesome little tool! I was using PhotonFileValidator but this is miles ahead of that one. I just got a new cheap creality ld002-r and the lack of tools is troublesome, it is like china is overtaking the market from the start with some working/cheap but also buggy solutions so it is inspiring to see some tools like this popping

**Describe the solution you'd like**
The island and resintraps are cool, but "possible overhangs" would be the most useful type of detection. Basically any "next/previous" big groups of pixels could be a possible weakpoint.

**Screenshots**
[image]

BTW the creality printer, for the price is surprising, I mean the firmware and chitubox has some bugs but nothing compared to my first fdm, that was a 2k USD makerbot!

sn4k3, 2020-09-23:
Cheap or expensive printers are independent from slicer result. If you use a bad slicer on a top printer the result will be bad no matter what. Also they must be tunned for each object, there's no 1 setting to rule-them-all

About overhangs i can implement a detection without fix for them. They must be deal on slicer, on PrusaSlicer you can configure the overhang threshold.

The objects you show on sample don't seen a problem to me on a real print, but a huge mass will

eried, 2020-09-23:
Well, as I said it is just my first days with the printer, so it is fine to use all ready to use stuff I can to start. If I start creating my own profile in PS I might just find problems that do not actually exist. And, yes, those areas wont produce any problem, I agree a bigger one will, I was just trying to provide a example.

sn4k3, 2020-09-23:
> If I start creating my own profile in PS I might just find problems that do not actually exist. 
Go UVtools -> Help -> Install profiles. Check if you printer is there

eried, 2020-09-23:
oh wow! nice. I will try it. On a super quick look, Chitubox supports seem to be a bit ahead prusaslicer

sn4k3, 2020-09-23:
That will change on next PS release. But slicing quality is better

sn4k3, 2020-09-30:
Implemented

closed 2020-09-30 by sn4k3

eried, 2020-09-30:
Hi, what is the concept behind this? I mean I am trying to understand how it works, for example if you check this:
https://www.screencast.com/t/UhWAGUD0

There is actually no overhang right? it is just a solid block but overhang is detected. 🤔

sn4k3, 2020-09-30:
That may be a bug, can you send me your file?

sn4k3, 2020-09-30:
I also have an ideia, chitubox new version are creating fade pixels, it seens like white but they arent.
Can you try run a 1 or 2 GapClosing iterations with Morph or layer repair?
Then redetect overhangs

sn4k3, 2020-09-30:
Just confirmed, is what i think.

[image]

Look that layer, one part of object is less white. Creating a overhang. I will threshold layer before to fix this

eried, 2020-09-30:
oh you are right. Here is the file just in case: https://mega.nz/file/q88UCYTK#EKir_LNycnywTTfznP1xlmBbADbQwRzRUHv1i74ucc4

I do not see a way to disable this in Chitubox, antialiasing is ON but on 0 by default. I know you recommended prusa slicer, but the supports are so nice on chitubox :/

sn4k3, 2020-09-30:
I already fix the issue, will be avaiable on next patch

eried, 2020-09-30:
nice! I am trying to print a big [case](https://www.thingiverse.com/thing:4260973) and using your tool for learning why each print has small defects (deformations on the result) is the only tool I was able to find

sn4k3, 2020-09-30:
deformations can be many variables out of UVtools scope, like LCD, LEDs, Z...

eried, 2020-09-30:
yeah I know, I am just using uvtools for trying to find the culprit. It is usually on the thin "overhangs" even if they are small

sn4k3, 2020-09-30:
BTW uvtools by default are set to find overhangs larger than 5mm (the 49px you saw on settings)