sn4k3/UVtools #65: [BUG] Blurry UI

eried, 2020-09-24:
**Describe the bug**
UI is blurry on high dpi configurations

**To Reproduce**
Open the app on any machine with a dpi setting different than 100%, i.e. any new 4k laptop (150% or 200% is the usual default) 

**Screenshots**
Up one is with high-dpi forced, low one is the blurry current one.
[image]

**Additional context**
The solution is not as simple as adding:
  ```
  <System.Windows.Forms.ApplicationConfigurationSection>
        <add key="DpiAwareness" value="PerMonitorV2" />
        <add key="EnableWindowsFormsHighDpiAutoResizing" value="true" />
    </System.Windows.Forms.ApplicationConfigurationSection>
```
to `app.config`, there are hardcoded sizes for the controls that dont scale properly :/

sn4k3, 2020-09-24:
yep that will happen when you scale windows or accept visual fixes from windows, thats a common problem with many apps.
I'm droping GUI development in favour of new GUI that support multi-os, is being rewritten and this problem will go away on that version

sn4k3, 2020-10-21:
Will fix on v1

closed 2020-10-21 by sn4k3