sn4k3/UVtools #66: [BUG] No shortcuts on programs/desktop when using UVtools_X.X.X.X.msi

eried, 2020-09-24:
**Describe the bug**
There is no icon created anywhere to access UVtools.

**To Reproduce**
Steps to reproduce the behavior:
1. Install UVtools_v0.8.2.4.msi
2. You need to execute "C:\Program Files (x86)\UVtools\UVtools.exe" directly

**Expected behavior**
Icon should be create on desktop and/or start menu

sn4k3, 2020-09-24:
For me is working, both menu and desktop shortcut are present

[image]

[image]

eried, 2020-09-24:
Oh weird, maybe is an older release? I tried in 3 machines. No icon created on any of them.

sn4k3, 2020-09-24:
I will check latter with diferent msi

eried, 2020-09-24:
I mean the icon was not removed after updating. You can see in Windows Sandbox that there is no new icon after you run the msi: 
[image]

sn4k3, 2020-09-24:
Cath the problem, will be fixed on a next patch

sn4k3, 2020-09-30:
Fixed

closed 2020-09-30 by sn4k3

eried, 2020-09-30:
Awesome, thanks!