sn4k3/UVtools #67: [QUESTION] How to use the GRAY2RGB and RBG2GRAY encoding for Bene Mono?

Klausxdy, 2020-09-29:
I want to convert profiles from grey to 1/3rgb but don't know how to use it in the UVtools. Also, UVtools can not import CWS profile and I don't know why. Here is an example CWS.
[_ResinXP2-ValidationMatrix_test_.zip](https://github.com/sn4k3/UVtools/files/5298924/_ResinXP2-ValidationMatrix_test_.zip)

sn4k3, 2020-09-29:
The .cws sample you gave is malformed.
.CWS inner files should have contents filename equal to file name

so instead of slice* they must be _ResinXP2-ValidationMatrix_test*
does nova able to print this file?

Heres an example of a working file:

[image]

Additional you cant convert grey to 1/3RGB, UVtools auto detect the output/input by image channels or flags that come from a SL1 file to make use of that. Currently there's no way to force a convertion.
If you are slicing your files you can use PrusaSlicer and use Bene printer, that will convert greys to rbg

Klausxdy, 2020-09-29:
Hi,

Thank you for your quick reply and help. I understand your first point and I will check it tonight with my printer.

For converting grey to 1/3RGB, is there a plan to add this function into UVtools? Because I like chitubox’s blur feature so I want to use chitubox to slice the model. It works well with bene4 color machine but can’t work with bene4 mono. (Lychee slicer as well, can’t slice model to 1/3rgb) I’m looking for a method to use chitubox to slice the model and then convert grey to 1/3RGB so I can print the model in bene4 mono.

Thank you again for your help.

Best Regards,

Duanyang Xu
Optoelectronics Research Centre
University of Southampton

[image]<https://eur03.safelinks.protection.outlook.com/?url=https%3A%2F%2Fuser-images.githubusercontent.com%2F113281%2F94572771-4ce79880-0269-11eb-8717-0377bd61900a.png&data=01%7C01%7Cdx1a17%40soton.ac.uk%7C5d4d2f5eff9a474d021a08d864859b3b%7C4a5378f929f44d3ebe89669d03ada9d8%7C0&sdata=Q%2Bpn%2FpbrPBfyumC0AGo6KwwlJH6gudwlZNAnJI7H%2FfI%3D&reserved=0>

sn4k3, 2020-09-29:
I didn't add a method to do that because you always need other parameters, for example print volume size, resolutions and other flags and comes from slicer and preconfigured to a machine. For example you can convert from sl1 epax10 to ctb epax1, it's converted but wont work because target machines are diferent. Same with your approach, i guess bene4 color not the same as bene4 mono in terms of sizes or resolutions, so a image convertion only is not sufficient.

PrusaSlicer allow me to add notes and them use them as flags. Chitubox don't have any field i can use other than namechine name. So if you want i can lookup for machine name and if equal to Bene4 Mono do the magic

Klausxdy, 2020-09-29:
Hi,

There is a plugin to slice .cws in chitubox. https://github.com/opensourcemanufacturing/Nova3D-Elfin-Chitubox-Plugin It can edit almost every parameters in the profile so it is simple to adjust the parameters to fit bene4 mono (I have done that). For info bene4 mono uses a 6.08’ mono screen with 1620*2560pixel (ZX608XI5249727).

Also about the converting, I’m using this https://github.com/MakerMatrix/PythonScripts/tree/master/monoMSLAConvert python codes to do the work and I made a small change to make the codes can process thousands slicer once. But is is still friendly to normal users without python environment.

#!/usr/bin/python3
#
# Copyright 2020 Photonsters
# Jarrod A. Smith (MakerMatrix)
#
# Permission is hereby granted, free of charge, to any person obtaining a copy
# of this software and associated documentation files (the "Software"), to deal
# in the Software without restriction, including without limitation the rights
# to use, copy, modify, merge, publish, distribute, sublicense, and/or sell
# copies of the Software, and to permit persons to whom the Software is
# furnished to do so, subject to the following conditions:
#
# The above copyright notice and this permission notice shall be included in all
# copies or substantial portions of the Software.
#
# THE SOFTWARE IS PROVIDED "AS IS", WITHOUT WARRANTY OF ANY KIND, EXPRESS OR
# IMPLIED, INCLUDING BUT NOT LIMITED TO THE WARRANTIES OF MERCHANTABILITY,
# FITNESS FOR A PARTICULAR PURPOSE AND NONINFRINGEMENT. IN NO EVENT SHALL THE
# AUTHORS OR COPYRIGHT HOLDERS BE LIABLE FOR ANY CLAIM, DAMAGES OR OTHER
# LIABILITY, WHETHER IN AN ACTION OF CONTRACT, TORT OR OTHERWISE, ARISING FROM,
# OUT OF OR IN CONNECTION WITH THE SOFTWARE OR THE USE OR OTHER DEALINGS IN THE
# SOFTWARE.

from PIL import Image
from os import listdir
from os.path import isfile, join
import sys
import os

# Parse the command line
try:
infile = sys.argv[1]
outfile = sys.argv[2]
except:
print ("Usage: " + sys.argv[0] + " <inFile> <outFile>")
exit(1)
if not os.path.isdir(infile):
    print("\n Infile is not a directory!")
    exit(1)
if os.path.exists(outfile):
    print("\n Outfile Already exists!")
    exit(1)
os.mkdir(outfile)
onlyfiles = [f for f in listdir(infile) if isfile(join(infile, f))]

for f in onlyfiles:

    filename = infile + "/" + f
    # Open the input image
    try:
        inImg = Image.open(filename)
    except:
        print( "ERROR: Could not open " + filename + " for reading.")
        continue

    # Get the input image type and decide which way we will convert it
    inType = inImg.mode
    if inType == "RGB" or inType == "RGBA":
        inImg = inImg.convert("RGB")
        converting = "C2G"
        print("Converting", inImg.size, "RGB-encoded grayscale image to full-resolution gray.")
    elif inType == "LA" or inType == "L":
        inImg = inImg.convert("L")
        converting = "G2C"
        print("Converting", inImg.size, "full-resolution grayscale image to RGB-encoded grays.")
    else:
        print( "You did not supply a suitable image.\nRGB, RGB/alpha, ")
        print( "8-bit grayscale or 8-bit grayscale/alpha are supported.")
        continue

    # Do an rgb-encoded to grayscale conversion
    if converting == "C2G":
        rgbPixels = inImg.load() # Create a pixel access object
        # Create empty grayscale output image with the converted X-dimension
        outImg = Image.new("L", (inImg.width*3, inImg.height))
        grayPixels = outImg.load() # Create a pixel object for converted output
        for x in range(inImg.width):  # Loop over the columns
            for y in range(outImg.height):  # Loop over the rows
                for channel in range(3):  # Loop over this pixel's R, G, B channels
                    # Build a gray pixel from the current channel's value
                    newPixel = (rgbPixels[x, y][channel])
                    # Assign the new gray pixel into the correct full-res index
                    grayPixels[((x*3)+channel), y] = newPixel
    # Do a grayscale to rgb-encoded conversion
    elif converting == "G2C":
        grayPixels = inImg.load() # Pixel access object
        # Create empty RGB output image with the converted X-dimension
        outImg = Image.new("RGB", (int(inImg.width/3), inImg.height)) #
        rgbPixels = outImg.load() # Create pixel object for converted output
        for x in range(outImg.width): # Loop over the output columns
            inX = x * 3 # The input X-dimension is 3x the output rgb X-dimension.
            for y in range(outImg.height):  # Loop over the rows
                # Build a new rgb-encoded pixel from the next three gray pixels
                newPixel = (grayPixels[inX, y],
                        grayPixels[inX+1, y],
                        grayPixels[inX+2, y])
                rgbPixels[x, y] = newPixel # Assign the rgb-encoded pixel

    output = outfile + "/" + f
    # Save the converted image
    try:
        outImg.save(output)
    except:
        print( "ERROR: Could not open " + output + " for writing.")
        continue

Duanyang Xu
Optoelectronics Research Centre
University of Southampton

sn4k3, 2020-09-29:
ok i will add a flag to chitubox files, you just need a profile with a printer named Bene4 Mono and it will do the convertion to rbg

Klausxdy, 2020-09-29:
Hi,

Thank you for your very kind help! Hope to see it soon.

Best regards,

Duanyang Xu
Optoelectronics Research Centre
University of Southampton

sn4k3, 2020-09-29:
So here the workflow:

You will need to configure a printer called Bene4 Mono, it must contain that words but also you can prepend or append other words as long Bene4 Mono is on string, eg: "My Bene4 Mono Printer" is valid as name
After you need to configure the printer sizes, those settings can be copied from: 
[image]

Then you slice to .ctb, go to uvtools and convert to .cws, it will produce the right images for mono

sn4k3, 2020-09-29:
Also if you only want the blur function, UVtools offer multiple blur technics

sn4k3, 2020-09-30:
Implemented

Klausxdy, 2020-09-30:
Dear,

Thank you for your quick solution! Is there a solution for Chitubox slicer not for PrusaSlicer? I’m not a PrusaSlicer user and it is not a common tool to slice models among my friends. For example using Chitubox to slice and then convert the slices from grey to 1/3rgb. I have a bene4 and a bene4 mono (anycubic mono x and phrozen mini 4K are also coming) so I’m happy to do some tests for you if you need.

Best regards,

Duanyang Xu
Optoelectronics Research Centre
University of Southampton

sn4k3, 2020-09-30:
The solution i gave is for Chitubox, please carefull read my comments

Klausxdy, 2020-09-30:
Sorry I just tested it with uvtools 0.8.2.4. I’ll test the new version now.

Duanyang Xu
Optoelectronics Research Centre
University of Southampton

Klausxdy, 2020-09-30:
Hi, sorry to bother you again. I have tested for serval times but it seems like not working. (I did Every step follow your suggested workflow) The converted .cws profile is still grey slices. Here are the .ctb from Chitubox (with the name of Bene4 Mono printer) and converted .cws from UVtools and also a .cws directly from nova3d bene's software Novamaker.
[final test.zip](https://github.com/sn4k3/UVtools/files/5302767/final.test.zip)

sn4k3, 2020-09-30:
You didnt proper set the machine name:
Look at UVtools status bar.

[image]

Post a print screen from your printer configuration at chitubox

Klausxdy, 2020-09-30:
[image]

sn4k3, 2020-09-30:
Can you export and attach your print profile here so i can import it?

Klausxdy, 2020-09-30:
[_Bene4_Mono_Profile_profile.zip](https://github.com/sn4k3/UVtools/files/5302801/_Bene4_Mono_Profile_profile.zip)

Klausxdy, 2020-09-30:
It seems like changing the name in Chitubox will not change the machine name in the profile.

sn4k3, 2020-09-30:
OK it seens machine name is only valid for inbuilt Chitubox printers and cant be changed.
I have to find another way to do this...

Klausxdy, 2020-09-30:
Anyway, thank you again. I'm going to sleep now (UK 3:30 AM). No rush and hopefully you can find another way to do this and I can test it on the machine. Good night!

sn4k3, 2020-10-01:
I have an ideia to overcome this, use the filename instead, so if name name contains bene4_mono it trigger the encoding/decoding.
eg: ResinXP2-ValidationMatrix_test_bene4_mono.ctb
Will implement this on next patch

sn4k3, 2020-10-01:
Implemented

closed 2020-10-01 by sn4k3

Klausxdy, 2020-10-01:
Thank you! I will try it tonight!

Klausxdy, 2020-10-05:
It works well! Thank you!