sn4k3/UVtools #68: [BUG] .cbddlp opened in chitubox swap x/y axes

tofof, 2020-09-30:
**Describe the bug**
Opening uvtool's cbddlp in chitubox displays with the x and y axes swapped. 
However, the files print correctly.
This bug has existed for a couple months at least, but 100% of my prints during that time work as expected, even though they display incorrectly in chitubox.

**To Reproduce**
Steps to reproduce the behavior:
1. Use Elegoo Mars printer in Prusaslicer
2. Export file as sl1
3. Use UVTools to convert sl1 to cbddlp
4. Open cbddlp in chitubox

**Expected behavior**
3d reconstruction in chitubox looks accurate, instead of swapping x/y axes.

**Screenshots**
[image]
[image]

**Files**
[test-swap-axes.zip](https://github.com/sn4k3/UVtools/files/5307711/test-swap-axes.zip)

**Desktop (please complete the following information):**
OS: Win 10 Pro x64 1803
UVTools 0.8.3.0 and earlier versions, through at least 0.4.2.1
Chitubox 1.7.0 and earlier versions, through at least 1.6.3

**Additional context**
I first noticed this right after chitubox had updated so I thought it was chitubox's fault, but today I confirmed that the problem existed even on uvtools 0.4.2.1 and chitubox 1.6.3 -- the first versions of uvtools and chitubox I used together. To do that, I uninstalled the elegoo mars printer, installed the printers from 0.4.2.1, redid the steps in 'to reproduce', and opened in chitubox 1.6.3 -- still had same behavior.

sn4k3, 2020-09-30:
Yes the 3D viewer will shows distorced, but layer mask must display ok. However in your image it doesnt look right.
Tools like uv3dp produce same output. Thats due PrusaSlicer cant be configured as normal i dont know why, but you need to invert XY for it to work when converting AND to display right. I need to dig into this to discover how to use PS without the FLIP_XY flag i use

tofof, 2020-09-30:
Yeah, if I remove the FLIP_XY from the printer and change "display orientation" to "landscape", it displays correctly in uvtools and in chitubox.
However, my printer refuses to print the file, saying an error about width/x not 2560 or something similar.

sn4k3, 2020-09-30:
Different softwares have different ways of build the slice, orientation on PS is also different from Chitubox, for example in PS you must use Mirror vertical instead of horizontal on your eleego and much others.
I have tried many combinations on PS to allow me to use it without flipping XY on code, but all attempts fails.
You could try combinate by yourself, as two pair of hands can be better.

The file on UVtools after convertion must respect your printer resolution, uvtools XY == chitubox printer XY or else it will fail.
Also orientation must be portait for that printer

tofof, 2020-09-30:
I think I got it.

tofof, 2020-09-30:
PrusaSlicer settings
Bed shape x=68.04 y=120.96
Display width 68.04
Display height 120.96
Number of pixels in x=1440 y=2560
Display orientation landscape
Mirror Vertically yes (same as uvtools elegoo mars printer setting)
Remove FlipXY from notes page

This produces
[image]
[image]
[image]
[image]
[image]

tofof, 2020-09-30:
Now the problem is how UVTools displays the image, though. 
It needs a way to reverse the effect of 'mirror vertically' setting to display correctly.

sn4k3, 2020-09-30:
UVtools display raw image, is not manipuled.
What about play with horizontal mirror?

tofof, 2020-09-30:
>UVtools display raw image, is not manipuled.
Well, yes and no. You already have a 'rotate' option.
The problem here is that chitubox format's raw data is as if you were looking at the layer from below, rather than from above. So there is no way to get the layer preview in UVTools and in Chitubox to agree unless you can mirror UVTools.

>What about play with horizontal mirror?
I tried. The actual printed object would end up backwards.

tofof, 2020-09-30:
Note that the mirroring is already a problem - UVTools with current settings has a mirrored preview AND has wrong xy in chitu.

Current
=====
https://i.imgur.com/7ONTaDq.png
https://i.imgur.com/7ONTaDq.png
https://i.imgur.com/TBkWmUu.png
https://i.imgur.com/CllOVIk.jpeg

My settings
========
https://i.imgur.com/lfT4Rim.png
https://i.imgur.com/882M9vb.png
https://i.imgur.com/QbmU9WW.png
https://i.imgur.com/TBkWmUu.png

sn4k3, 2020-09-30:
> Well, yes and no. You already have a 'rotate' option.

Disable auto rotation to not confuse the comparing here

> Note that the mirroring is already a problem - UVTools with current settings has a mirrored preview AND has wrong xy in chitu.

Not UVtools, its prusaslicer, if you open the sl1 zip file you will notice images are output like that.
Have you tried multiple combinations with Mirror horizontal and/or/not vertical on PS?

I already kwen origin point was diferent from PS and CB

Also are the images you post on Chitubox the original or converted after uvtools?

tofof, 2020-09-30:
My workflow was PS -> UVT -> Chitu. The images in chitu are of files produced by UVTools.

> Not UVtools, its prusaslicer, if you open the sl1 zip file you will notice images are output like that.

Right, the images are what the printer actually puts on the UV screen, so they have to be mirrored.

> Have you tried multiple combinations with Mirror horizontal and/or/not vertical on PS?

Yes, I have tried all 4 ways. The only one that gives correct output is the one I specified.

tofof, 2020-09-30:
So at this point there are 2 different issues.

1) using FlipXY causes problems with Chitubox.
First problem: distorted geometry from pixels being scaled incorrectly according to backwards aspect ratio
Second problem: voxels are placed outside the bounds of the plate. 
Because of the second problem, you cannot make edits inside chitubox and re-save the file. 

This problem is completely fixable by switching to my printer settings.

2) UVTools always displays the layers as they appear as raw images inside sl1 files. 
This means that they appear mirrored if you mentally consider yourself to be looking down at them.

This problem is independent and is present no matter what settings are used for the printer.*

I should have made the suggestion to include a mirror display option in UVTools as a second issue, because it confused this one.

*It is possible to change mirroring settings so that UVTools looks correct, but doing that will always mean that the actual print will be mirrored.

sn4k3, 2020-09-30:
Something is wrong here, for example: 

> bed x68.04, y120.96, pixels x2560 y1440, landscape, yes horizontal mirror, no vertical mirror, no flipXY (WORKS, but will look mirrored in UVT)
> bed x68.04, y120.96, pixels x2560 y1440, landscape, no horizontal mirror, yes vertical mirror, no flipXY (WORKS, but will look mirrored in UVT)
> bed x68.04, y120.96, pixels x2560 y1440, landscape, yes horizontal mirror, yes vertical mirror, no flipXY (object will print mirrored)

How that will work without FLIP_XY when you use x2560 y1440?
To fix this we must configure PS to match chitubox: bed x68.04, y120.96, pixels x1440 y2560

> Issue 1 - using FlipXY causes problems with Chitubox.

That's not FLIPXY fault, is because i only flip resolutions, i still need to flip display width and height to also match chitubox, that's what cause wrong preview for sure

tofof, 2020-09-30:
> Something is wrong here, for example:

Yes, I skipped a row when writing them up in github -- and then continued using old data on clipboard not realizing I'd skipped.  Fixed above.

You can discard the hack by changing the bed size in Prusa.
Elegoo Mars is x1440 y2560 according to manufacturer preset in chitubox.
**X is the short dimension.**
Current UVTools printer settings have bed size x=120.96 y=68.04.  X is incorrectly the long dimension. This is the main problem.

sn4k3, 2020-09-30:
Flip display fixed the preview problem

[image]

tofof, 2020-09-30:
You need to hit 'back' button and make sure that it is not drawing out of bounds, too.
That is what my post https://github.com/sn4k3/UVtools/issues/68#issuecomment-701591860 was about.

sn4k3, 2020-09-30:
> You can discard the hack by changing the bed size in Prusa.
> Elegoo Mars is x1440 y2560 according to manufacturer preset in chitubox.
> X is the short dimension.
> Current UVTools printer settings have bed size x=120.96 y=68.04. X is incorrectly the long dimension. This is the main problem.

I know, i do this because otherwise when converting it produce wrong data as PS output bad images if you put the right resolutions and widths

tofof, 2020-09-30:
But it doesn't.
[right dimensions.zip](https://github.com/sn4k3/UVtools/files/5308561/right.dimensions.zip)

sn4k3, 2020-09-30:
We might look at diferent things. Can you send me a sliced 3mf file so i can review all settings.
Any way with my fix on uvtools code this is fixed as it is now

tofof, 2020-09-30:
[top LR.zip](https://github.com/sn4k3/UVtools/files/5308571/top.LR.zip)
The 3mf that made those.

sn4k3, 2020-09-30:
Ok that worked because of Landscape and thats why it come mirroed, those printers are landscape, with that we are "hacking" prusaslicer creating a portait image but in landspace mode, so things come mirroed.
I'm gona publish a patch here for you try with stock uvtools settings and printers

sn4k3, 2020-09-30:
https://we.tl/t-Z0VlWJIGU3

try that build and report back, use default uvtools printers with flipxy and check chitubox preview

tofof, 2020-09-30:
Yes, that build works.
Using the printer installed from that build, conversion to cbddlp produces a file that displays correctly in chitubox.

tofof, 2020-09-30:
I'm glad you found the bug with display width.

Can you explain the advantage of deliberately setting X/Y backward in PS and then using FlipXY, instead of setting them as I do in the 3mf I sent?

sn4k3, 2020-09-30:
> Can you explain the advantage of deliberately setting X/Y backward opposite to ChituBox in PS and then using FlipXY? It seems like setting them to match (as in the 3mf I provided) would be a second win.

Because FLIP_XY flag was the only way i found to make it match chitubox and respect the portait mode.
Using correct values on PS with landscape as you are doing flips de image:

Left: FLIP_XY stock printer | Right: Correct values, no FLIP_XY and landscape
[image]

tofof, 2020-09-30:
I don't understand. 
The left image thumbnail has the tail at the left. The layer preview has the tail at the right. These seem 'flipped'.
The right image thumbnail has the tail at the left, and the layer preview has the tail at the left. These seem 'matched'.

sn4k3, 2020-10-01:
Ignore thumbnails slicers can generate those random, with logic or whatever, what counts here is the masked image, so ignore that.
I'm performing more tests to find if i can go without FLIP_XY

tofof, 2020-10-01:
Importantly, it's not mirrored, it's rotated 180. (Not sure which you meant by "flips").

sn4k3, 2020-10-01:
[image]
[image]

That's all the possible combinations, see titles to know which combinations i used, first picture is stock uvtool printer

sn4k3, 2020-10-01:
From all landscape combinations, none was able to produce same image as stock with portait

tofof, 2020-10-01:
Question.
The shape is just /different/ in the rotations vs the first image.
It seems like the rightmost of the top 4 should be identical but that it's a different arrangement of support pads.

sn4k3, 2020-10-01:
The file was the same on all, i didnt even close prusa slicer, just changed settings from screenshot to screenshot

tofof, 2020-10-01:
Do you see what I mean though?
Is that what you're trying to show? That it generates a completely different support structure?

tofof, 2020-10-01:
I don't mean different aspect. I mean the actual shape is different. 
There are two notches near the top in the first image, and not in the rest.

tofof, 2020-10-01:
[image]

sn4k3, 2020-10-01:
> I don't mean different aspect. I mean the actual shape is different.
> There are two notches near the top in the first image, and not in the rest.

But stl is the same xD

and btw, if i slice same stl on chitubox i got this:
[image]

So it look like match with Landscape NoMirrors?

tofof, 2020-10-01:
But we're not looking at STL, we're looking at the support pads generated on layer 0. No part of that is the smooth-sided shape from the STL. 
Set pads and supports to 'none' in prusa.

sn4k3, 2020-10-01:
Ok i will import stl with pad already one minute

sn4k3, 2020-10-01:
Here the new collection, pads and supports off. Stl with pad inbuilt

[image]

tofof, 2020-10-01:
Did you actually disable the pad and support settings in prusa? 
[image]
Or did it generate its own supports underneath the supported stl?

Once again, the 1st image doesn't match.
The first image has a notch at bottom center, the other 4 are identical to one another (in various rotations / mirrorings) but all lack that notch.

sn4k3, 2020-10-01:
[image]

tofof, 2020-10-01:
Agreed, it matches landscape mirror vertical.

sn4k3, 2020-10-01:
> Agreed, it matches landscape mirror vertical.

Ok we are getting somewhere here.
If i import same STL on chitubox the result will match PS landscape NoMirrors.

Can you confirm this on your machine?

sn4k3, 2020-10-01:
Ok i think i got it.
Chitubox preview is not the actual image that goes to printer, in fact after open file with uvtool image is mirroed cause the setting on chitubox to mirror.
So in end it will match with landscape mirror horizontal.
So i think its safe to get rid of FLIP XY and convert everything to this pattern, and use MoMirrors when not mirroing on CB and horizontal when do LCD Mirror on CB

[image]

tofof, 2020-10-01:
STL sliced by chitubox 1.7.0 matches PS Landscape HorizontalMirror.
Are you sure you're not looking at the preview inside chitubox itself? Chitubox mirrors the preview, for the exact reason I want you to mirror the display in UVTools.

sn4k3, 2020-10-01:
So i'm gona convert all printers to this mode. and release on next patch. FLIP_XY flag will keep for compability on older profiles.
About fliping view i can put a flipped virtual image as i do for rotate but that will complicate everything as it require translated pixels when drawing over a manipulated view, the click point must match with original image

tofof, 2020-10-01:
Yeah, flipped virtual image is all I was hoping for on that issue. It would make going between PS and UVT much easier.

tofof, 2020-10-01:
👍 on converting printers to this mode, I think having everything agree what's x and y is useful

sn4k3, 2020-10-01:
> Yeah, flipped virtual image is all I was hoping for on that issue. It would make going between PS and UVT much easier.

I can attemp this latter on a future release, it will require a full code review on all operations over image, if fliped i have to unflip on all translation logic. 
Day by day it grow stronger ^^

tofof, 2020-10-01:
Yes, you would have to be very careful on all the drawing/ROI operations (and I suppose it means mirroring the  kernels for the other operations, since user could put in one that's not symmetric).
I can help bugtest that feature whenever you decide to attempt it.

tofof, 2020-10-01:
Anyway great work, as usual for you.

closed 2020-10-01 by tofof