sn4k3/UVtools #70: [FEATURE REQUEST] The documentation is incomplete for Linux and Mac

brandon-fryslie, 2020-10-10:
**Describe the bug**
The documentation for non-Windows operating systems is incomplete.  This program looks awesome and I'd love to use it, but unfortunately I can't figure out how to get it running on Mac/Linux.  I ran `UVTools.sh` but I get this error:
```
Could not execute because the specified command or file was not found.
Possible reasons for this include:
  * You misspelled a built-in dotnet command.
  * You intended to execute a .NET Core program, but dotnet-UVtools.dll does not exist.
  * You intended to run a global tool, but a dotnet-prefixed executable with this name could not be found on the PATH.
```

Maybe there are missing steps in the documentation that would allow me to run the application?  Thank you!

sn4k3, 2020-10-10:
Yes that's true for Mac but not for linux. 
That instructions are for the future UVtools, not the current release.
You can test a preview of upcomming uvtools here: https://easyupload.io/vtvmqk

I should have put a note on that and probally will now

brandon-fryslie, 2020-10-12:
OK, thank you!  I got it running no problem :) Thanks for the great program!

closed 2020-10-12 by brandon-fryslie

sn4k3, 2020-10-12:
Keep an eye on uvtools group to get the newest's preview releases, gona send a new one in a day or two for testing