sn4k3/UVtools #71: [BUG] Islands detected when there is no island

eried, 2020-10-11:
I think this is because the gray pixels from Chitubox, like before. But I tried playing with the thresholds without success

[image]

sn4k3, 2020-10-11:
That is maybe a resin trap, selected island shows 16 pixel, but on view is way more pixels. Please confirm that, otherwise send me the file for inspection

eried, 2020-10-11:
Oh you are right! I was wondering why the islands were not repaired, but it was because the next layer was "detected" after the repair so I was just confused with resin traps.

eried, 2020-10-11:
(Thanks) 

question -> any reason to limit the "removed island max pixels" numericupdown to 255?

closed 2020-10-11 by eried

sn4k3, 2020-10-11:
Islands are Yellow, traps are orange but you can configure colors to avoid this.
Also you can hold CTRL and click on a issue on image to go to it on list
About your question, 255 is already a high value of pixels to remove, that is a big mass and should be supported instead of removed, 255px is +/- 13mm and so is dangerous to process automatic. If you want to remove big masses manual is the way to go