sn4k3/UVtools #72: [FEATURE REQUEST]

eried, 2020-10-11:
Since an island could be part of an overhanged feature, would not be logic to have a checkbox below "Repair islands" / "Recursive repair" that just repeats the repair process until no more issues are removed? (just a very simple loop)

sn4k3, 2020-10-11:
You are asking a recursive function for the remove islands threshold right?

eried, 2020-10-11:
I think. If there is a feature hanging and "floating", UVtools will detects the first part as issue, after you fix that, the next layer will be marked with the issue now, until you get to the "root", effectively removing the whole feature that was hanging right?

I think it would add value to just iterate removing islands until no more islands remain. What you think?

sn4k3, 2020-10-11:
That can be done for the removal function, however is always a risk, if you have recursive lefover islands in most of cases that are required to model and worth to support. But i can add that and let user decide

eried, 2020-10-11:
That might be true, but since you have the area limit in pixels it will always stop before removing big stuff, except ofc if the feature itself is like a tentacle or hair 🤔

sn4k3, 2020-10-11:
Implemented, will be avaliable on next patch

[image]

eried, 2020-10-11:
Very awesome! I will try once is released. I fixed some hand-sculpted figure and it required a lot of cleanup just running the ->detect-> fix issues loop, so it is the perfect test

Have you thought on making the reading and issue finding/fixing async? like in a separate thread

sn4k3, 2020-10-11:
Operations are all made in a separate thread in parallel.
On this recursive function only layer issue removal is parallelized because next layer require the current layer been fixed first in order to detect the island, so it can't run in a parallel way under a layer range, that's why it will be slow

eried, 2020-10-11:
I see, but I mean without using a modal progressbar in the UI. i.e. when loading or when searching issues

sn4k3, 2020-10-11:
I can allow layer navigation while progress is going, but every other function will be locked or in best case be queued
Also update/display while got the results will slow down the processing by much.

sn4k3, 2020-10-13:
Implemented

closed 2020-10-13 by sn4k3