sn4k3/UVtools #73: [BUG] Circle overlay in pixel editor is weird

eried, 2020-10-11:
**Describe the bug**
The overlay on mouseover is weird if the thickness of the circle is not -1 and the circle has a large diameter

**To Reproduce**
Steps to reproduce the behavior:
1. Enable pixel editor, Go to pixel editor tab
2. Set:
[image]

3. Mouse over in the layer holding shift
4. The mouse over is not a circle, but a square with a circle combined

The pixels are modified correctly, it is just the overlay on mouse over

sn4k3, 2020-10-11:
That's because thinkness is draw from outside and inside model, meaning it will increase bounds. 
Cursor sets to brush size but when using thickness i should add brushsize + thickness/2

sn4k3, 2020-11-26:
Circles are still cutting at some point, adjusting cause bad aligment with the result position. Leaving as it is for now.

closed 2020-11-26 by sn4k3