sn4k3/UVtools #74: [BUG]

Angelus1981, 2020-10-12:
**Describe the bug**
After using the repair tool I have quite a lot issues that within a support there are layers without pixels.

**To Reproduce**
I use Chitubox 1.7.0 as slicer for my Elegoo Saturn. The   file is saved as .ctb file.

**Expected behavior**
There shouldn't be empty layers within a support after using the repair function

**Screenshots**

**Files**
I do not know how to upload files to Github but I can provide a file where I have the issue

**Desktop (please complete the following information):**
 - OS: Win 10 64 bit

sn4k3, 2020-10-12:
Most likely that are caused by wrong configuration under your specific use.
Can you post a screenshot of you tool repair window with advanced options showing?

Angelus1981, 2020-10-16:
Hi, Sorry for the late reply.

please find the screenshot attached. I only changed one value since this one seems to be too high for my support settings that I typically use without problems. But I had the described issue also with the standard settings that come with the installation.
[image]

Please find also 3 screenshots of the issue with the posted settings. 

Before using the fixing functions e.g. it shows 4 issues in layer 936 as seen in screenshot:
[image]

After fixing 1 issue is shown on a completely different location where suddently one layer for a support is emtpy. I made a picture of the layer mentioned in the issue list:
[image]

ans the layer below which is emtpy, causing the issue.
[image]

So before using the issue fixing tool layer 935 didn't had any issues there. Hope that helps.

sn4k3, 2020-10-16:
Ok so the title and main writing of your problem is wrong and lead me in error. 
You were refering empty layers, that is layers with 0 white pixels in total (all black), and that's not your case.
I've edit your title to correct the real discuss here.

UVtools never create new masses on repair, at maximum he increase brightness of already existing pixels or expand to bound two joins.
To confirm this open the file without repair and go to that region were you have that "new" island, hover with mouse using Control, look on bottom status bar and see the "B:" value, if anything greater than 0, than thats the problem. 
Please confirm this.

Also if you want attach your file to i check, via wetransfer for example

Angelus1981, 2020-10-16:
Sorry if heading was misleading. 
I tried to do what you wrote but I think it is fine that the B value is higher than 0 since this means there are pixels which sould be since there should be the support.

I now proceeded with version v8.5.0 and face the same issue but surprisingly on a different layer now in the same model.

So opened the file again and checked layer 949 and 948/947. There is no island since the support is available in both layers.
[image]
[image]
[image]

I run the "issues" operation to identify issues. The tool identifies the pixels from the support as error (pixels in layer 948 still there)
[image]
[image]

After repairing with the default settings from clean installation there is suddenly no pixels in the area of the support in layer 948 and some layers below (not the complete layer is empty only the region of the support). So it seems the tool deleted pixels during repair operation.

[image]
[image]
[image]

Settings (default as per installation):
[image]
[image]

I also sent you the file to the email given in your profile via wetransfer. Please have a look.

sn4k3, 2020-10-16:
Ok your problem are related to settings.

It is a island because bellow is unsafed supported, please read this thread: https://www.facebook.com/groups/uvtools/permalink/341571073619069?comment_id=342323286877181

About islands got remove is due new function, recursive removal, it remove islands below x pixel count, so when you run repair it will remove this island, then check upstair and verify same condition up remove everything that match that criteria.
If you want to disable this, set Revome islands equal or smaller than: 0 px.

UVtools is a generic program so the defaults will never match everyone, you must tune it for your use. Take your time to read the tooltips that are present on functions / labels / boxes.

Also keep in mind you dont have to clear all islands, a issue doesn't mean a real problem, it's more a warning that you must deal with. Try to clean all issues or lower safe values on uvtools is a bad pratice, you should decide if remove/fix a island is really needed.
Again: Just because you have 1 pixel supporting a bunch more, that doesnt mean is well supported, so UVtools will always report them as unsafe

Angelus1981, 2020-10-16:
Ok I will check with the next file that I will work on. I will give feedback ASAP

sn4k3, 2020-10-25:
Settings based

closed 2020-10-25 by sn4k3