sn4k3/UVtools #75: [BUG] No demo file

eried, 2020-10-14:
**Describe the bug**
Not loading the demo file

**To Reproduce**
1. Install latest
2. Open without file

Program tries to open the demo file from system32:
[image]

eried, 2020-10-14:
(is not that I think is important, but I saw it was on the release notes)

sn4k3, 2020-10-14:
How do you open uvtools? run as administrator?

eried, 2020-10-14:
No, but if I do, the result is the same.

sn4k3, 2020-10-14:
I wonder why you pc starts uvtools on system folder...
But i will set the read file from assembly directory rather than partial

eried, 2020-10-14:
There is no "start in" in the .lnk created by the .msi btw

sn4k3, 2020-10-14:
Test by start uvtools from .exe on it own directory as test

eried, 2020-10-14:
If I go to program files/ and execute the exe it works

sn4k3, 2020-10-14:
On shortcut you can place the working directory correctly.
I wil ltry find a way to auto set it via setup and update the MSI

sn4k3, 2020-10-14:
I reuploaded the fixed MSI.
Thanks for reporting

closed 2020-10-14 by sn4k3