sn4k3/UVtools #77: [BUG] Glitches/typos on 1.0

eried, 2020-10-22:
Cool that the new version supports dpi resize, it is very slow to launch (1s before vs 6-7s), but I guess that is ok, however some letters are disappearing randomly:

[image]

Highlight makes the properties invisible, and some dialogs are just empty:
[image]

Also there is some minor typos (issues and crosshair) and it is not very clear which filters are actually active (it is just two gray tones):
[image]
[image]

Windows 10, version 2004

sn4k3, 2020-10-22:
This GUI framework have some bugs and i'm using a preview version of it. I will try to forward that issue to them.
Can you try to put windows at normal view settings to see if it helps or fix any of the problems?
I will fix the typos.

eried, 2020-10-22:
This is on 100% scale
[image]

I can try later on my laptop.

sn4k3, 2020-10-22:
ok report back once you try on other system

eried, 2020-10-22:
In my laptop, which has an intel crap gpu, it works fine on any dpi. So it might be the graphics accelerator... strange

[image]

sn4k3, 2020-10-22:
Can you report the full system specs of the problematic system?

eried, 2020-10-22:
[image]

[image]

sn4k3, 2020-10-22:
well the specs seens solid...
I guess you use the dedicated gfx on system right? No hybrid combination nor shared for windows?

eried, 2020-10-22:
Yeah it is an old but dedicated card driving 4 screens. I will do more tests tomorrow, but this is first time I see glitches like this happening (missing letters)

sn4k3, 2020-10-22:
Can you try only with 1 screen?
This framework is not commun to find in a app, its a new thing and beta, it still have a long path, but they are in the right direction and the framework is strong

eried, 2020-10-23:
Sometimes it can look very artistic :)
[image]

I tried few settings in the nvidia control panel, disabling aliasing, etc, compatibility mode but none seem to fix the issue. But I do not need latest version in the office and it works fine in my laptop, so it is all good until that framework fixes this.

And I will post if I test one monitor, it will mess my layout if I disable the other screens.

sn4k3, 2020-11-01:
Can you try with the new version and report back? I upgraded framework version...

eried, 2020-11-01:
1.1.1 starts the process, without any window and quits immediately. Not sure if there is any log I can send, I tried also uninstalling it. I am guessing is this:
[image]

Im using 0.8.6 in both pcs, new non blurry ui is quite laggy on the laptop (not glitchy).

sn4k3, 2020-11-01:
Dam, i never expected  that winrt to be used! At least on my computer it works well without it.
I've fixed the MSI. You can redownload

sn4k3, 2020-11-12:
The text glitches still happen?

eried, 2020-11-12:
I don't see glitches on latest :) 

I noticed that the size of the font is big compared to the OS fonts so the toolbar looks a bit strange:
[image]

Also every couple of openings, it just quits once without any warning:
https://www.screencast.com/t/tkgc68Fgc

sn4k3, 2020-11-12:
You can decrease the font size of your program, you need to edit Assets/Themes/UVtoolsLight.xaml
[image]

Stange thing at quits, can you see the problem on windows events?

eried, 2020-11-12:
The crash in eventvwr is 
```

Fault bucket 1798254039142380948, type 4
Event Name: APPCRASH
Response: Not available
Cab Id: 0

Problem signature:
P1: UVtools.exe
P2: 1.2.1.0
P3: 5f3ed523
P4: coreclr.dll
P5: 4.700.20.41105
P6: 5f3397ec
P7: c0000005
P8: 0000000000191dcb
P9: 
P10: 

Attached files:
\\?\C:\ProgramData\Microsoft\Windows\WER\Temp\WER8BC9.tmp.WERInternalMetadata.xml

These files may be available here:
\\?\C:\ProgramData\Microsoft\Windows\WER\ReportArchive\AppCrash_UVtools.exe_8c6831b1a47ca43fc6766f682c74dcc9dd13ad_c0835d6b_b5c3af73-d9c1-440e-9882-cbaae04cb688

Analysis symbol: 
Rechecking for solution: 0
Report Id: 78e2c0fd-3014-45c6-bed3-d5443c5b641e
Report Status: 268435456
Hashed bucket: 2a7bcd6f37d89fcf08f4ae8562fe3594

```

eried, 2020-11-12:
And there is scaling but I have not seen any laptop with "around" 4k screen without scaling enabled by default.

sn4k3, 2020-11-12:
The scaling is horrible, read this: https://www.facebook.com/groups/uvtools/permalink/361211911654985/

eried, 2020-11-13:
I do not think is a gimmick, the text and images look sharper on high resolution screens, including small screens. However, old apps may look blurry but "High DPI properties" make them look sharp, i.e. 2007's Bome Restorator
[image]

You can see how that window and toolbar icons will look "fine" on a 1080p monitor at the same size (non-scaled vs scaled), but then PS1 games looked fine on a ntsc CRT :)

The DPI setting fixes all apps ... except UV tools; for some mysterious reason:
[image]

The old app only resize the text in that mode, and the new one resizes everything in a weird way. It might be just some manifest missing, or a property of the controls.

sn4k3, 2020-11-13:
It can look sharper because of higher DPI, but it depend on quality of the panel you get, there are FullHD and QHD with much sharper image then cheap 4Ks, but the tradeoff is big. You can't divide or multiply pixels on the LCD, meaning the Scale will repeat same pixel for a amount of XY, you are losing resolution and quality, get blury on some apps, and also many application dont care about scaling. For example, to constrain a window inside the monitor area the most people will use screen resolution only, so when we constrain window to be 2440px wide for example this is true at 100% scale. At 150% your system are still reporting 2440px at ResolutionX but now this is not true anymore because now you dont have 2440px usable area but 25% of that instead, so the window will overflow. So a 4K at 150% is a QHD view area.
Just like msla LCDs, two same lcds with same resolutions but the pixel size is different, you will get sharper quality with lower pixel size.
On PCs with scaled images you only take full advantage of it when playing games or movies in full screen, native resolution will be used at 100% at full screen GPU render. 

> The old app only resize the text in that mode, and the new one resizes everything in a weird way. It might be just some manifest missing, or a property of the controls.

New UVtools uses a diferent GUI framework that is scalable, that means everything will scale, if you use 150%, the sizes will be: size*1.5, and that is the real purpose of the scaling, get everything bigger. There still unfinish work or bugs on framework, as you can see on other issues, scaling is hidding buttons and breaking the layout, as i said before is always a curse for developers, theres nothing better than a clean native image.

sn4k3, 2020-11-13:
Here what i do to keep windows inside bounds for scaled images:

`ScrollViewerMaxHeight = Screens.Primary.WorkingArea.Height / Screens.Primary.PixelDensity - 300;`

**PixelDensity** here is just the SO Scale setting, 1 == 100%

eried, 2020-11-13:
Besides personal preferences, scaling is already enabled on all new laptops I have seen on the last decade. I was checking some stuff and I noticed if I reset all fonts to the Form editor defaults and remove this scaling code:
[image]

The size is perfect on 100% and higher (on higher dpi seems to require better icons):
[image]

sn4k3, 2020-11-13:
> Besides personal preferences, scaling is already enabled on all new laptops I have seen on the last decade. 

Yes, and it must be scaled, windows itself finds and sets the best scale value every fresh install, he know user wont able to read on current ratio, so they must come with this software solution, but IMO hardware problem should be fixed in hardware, software is just a patch/filter taking performance.
4K in such small laptops LCD size theres no way it will have good reading at 100%, again this is all a marketing BS. They put 4K to give an impression of "premium" or advantage over lower resolutions, but in fact they sell snake oil. You pay more but get worse. This also can happen on some FullHD and QHD resolutions not only 4K, the panel quality and DPI make a lot of diference.

> The size is perfect on 100% and higher (on higher dpi seems to require better icons):

That code was to fix the low font size on every form without manual set and forget something, it was based on a 100% scale
Icons got scaled too, as any image you know if you resize image you lose quality.
There also properties on .NET to resize on font or DPI, old UVtools was set to Font

sn4k3, 2020-11-19:
Main problem is fixed, closing this

closed 2020-11-19 by sn4k3