sn4k3/UVtools #78: [BUG] Minor 1.0/1.0.0.1 issues

tofof, 2020-10-23:
**Describe the bug**
Main interface buttons at top left: disabled buttons have no tooltip.  (Information, ????, Settings, ????, Log)
Pixel editor icon button is disabled until you click pixel editor tab/button above layer preview. Clicking icon should enable pixel editor. Tab/button above layer preview is probably not useful at all anymore.
Settings->Issues->"Pixel intensity threshold" defaults to 0, but can't be set back to 0 after change (minimum is 1).
Settings->Issues->"Supporting safe pixels..." is present twice.
Settings->Layer repair->Empty layers / Resin traps texts are swapped in the settings window. The 2nd setting changes the 2nd option in layer repair. The text in the tool window is correct (empty layer checked -> empty layers removed during repair). See image.

[image]

sn4k3, 2020-10-23:
> 1. Main interface buttons at top left: disabled buttons have no tooltip. (Information, ????, Settings, ????, Log)

They have, but disabled components never show tooltip. They should be not visible util they need to, but framework has a bug where prevent me to reshow the tab if i hidden them one time, so for now they must be disabled rather than hidden util they fix this part. That are for GCode and Pixel Editor.

> 2. Pixel editor icon button is disabled until you click pixel editor tab/button above layer preview. Clicking icon should enable pixel editor. Tab/button above layer preview is probably not useful at all anymore.

Thats related to 1. My ideia is it to behave like old version of uvtools. Waiting for a fix from framework.

3./4./5. Will fix on next patch

tofof, 2020-10-23:
1) Unfortunate but ok. What is the second box (```</>```)?
2) I was mistaken anyway about the function of the two buttons; I see now it's exactly like old uvtools.

And I forgot to say -- congratulations on 1.0 and nice new icon! 🎉

sn4k3, 2020-10-23:
1. GCode, they are the same function and in order as old UVtools, maybe you are aware of it because you dont slice files with present gcode so old uvtools never had a chance to show you that tab :) 

Thanks 👍 The icon is a bit old fashioned but my design skills are low/bad so i come with this simple logo

closed 2020-10-23 by sn4k3