sn4k3/UVtools #82: [BUG] UVTools zip release is packaged with backlash instead of slash

yaqwsx, 2020-10-26:
**Describe the bug**

UVTools v1.0.0.2 ZIP file downloaded from the release contain backslashes `\` instead of slashes `/` for directory separation in file paths. This causes trouble on MacOS and Linux: no directories appear. The [Zip specification](https://pkware.cachefly.net/webdocs/casestudies/APPNOTE.TXT) (section 4.4.17) specifies that the forward slashes should be used.

**To Reproduce**
Steps to reproduce the behavior:
1. Download the v1.0.0.2 release
2. Unzip it with Gnome Archive Manager
3. run either `sh UVtools.sh` or `dotnet UVtools.dll`
4. See error

```
Error:
  An assembly specified in the application dependencies manifest (UVtools.deps.json) was not found:
    package: 'System.Drawing.Common', version: '4.5.0'
    path: 'runtimes/unix/lib/netcoreapp2.0/System.Drawing.Common.dll'
```

Since there are no directories, only files with backlashes in names.

**Desktop (please complete the following information):**
```
Distributor ID:	Ubuntu
Description:	Ubuntu 20.04.1 LTS
Release:	20.04
Codename:	focal
```

sn4k3, 2020-10-26:
I don't understand why this is happening and only on that file... A Mac user had the sample problem, but i tested on my mac and linux test machines and for me works without any problem.
I'm using  Windows native zip but i can try use winrar or 7zip to do the packages

[image]

sn4k3, 2020-10-26:
I've updated the release zip, can you redownload and try?

yaqwsx, 2020-10-26:
When you use `unzip` on Linux, you get directories - `unzip` replaces backslashes with slashes. However, most of the GUI tools (e.g., Gnome Archive Manager) do not. I think it is better to follow standard practices (and the ZIP format specification) to avoid unpleasant surprises.

I tried the new file and works nicely. Thanks!

closed 2020-10-26 by yaqwsx