sn4k3/UVtools #85: delay time after first 10 slices on cws convert[BUG]

chris666uk1, 2020-10-29:
delay time after first 10 slices on cws convert 

get a 62 second delay after the first 10 slices on cws convert

[novamaker.txt](https://github.com/sn4k3/UVtools/files/5460182/novamaker.txt)
[converted.txt](https://github.com/sn4k3/UVtools/files/5460183/converted.txt)

i dont no if its me but everything looks good in prusa

sn4k3, 2020-10-29:
Can you attach your .cws file?

chris666uk1, 2020-10-29:
sure CWS and sl1 had to zip them

[SL1 Cat_t_Tough0.035mm_SL1_2h48m.zip](https://github.com/sn4k3/UVtools/files/5460376/SL1.Cat_t_Tough0.035mm_SL1_2h48m.zip)
[CWS Cat_t_Tough0.035mm_SL1_2h48m.zip](https://github.com/sn4k3/UVtools/files/5460378/CWS.Cat_t_Tough0.035mm_SL1_2h48m.zip)

sn4k3, 2020-10-29:
Found the problem, will fix on next patch.

The fix should produce the following:

**From:**

> ;<Slice> 2
> M106 S255
> ;<Delay> 15000
> M106 S0
> ;<Slice> Blank
> G1 Z4 F120
> G1 Z-3.96 F120
> ;<Delay> 72000
> 
> ;<Slice> 3
> M106 S255
> ;<Delay> 2500
> M106 S0
> ;<Slice> Blank
> G1 Z4 F120
> G1 Z-3.97 F120
> ;<Delay> 62500

**TO:** 

> ;<Slice> 2
> M106 S255
> ;<Delay> 15000
> M106 S0
> ;<Slice> Blank
> G1 Z4 F120
> G1 Z-3.96 F120
> ;<Delay> 15000
> 
> ;<Slice> 3
> M106 S255
> ;<Delay> 2500
> M106 S0
> ;<Slice> Blank
> G1 Z4 F120
> G1 Z-3.97 F120
> ;<Delay> 4500

To fix your file with future release you can open converted file, go GCode tab, hit Update GCode button and resave the file.
Right now you can do a search and replace on file and replace all 62500 with 5000 or 7000

sn4k3, 2020-11-01:
fixed

closed 2020-11-01 by sn4k3