sn4k3/UVtools #86: [FEATURE REQUEST] Show user the changes they've made

mylife4aiurr, 2020-10-31:
So for instance you've made multiple changes using uvtools.

I morph erode bottom layers to over come elephants - use more erosion
I morph erode layers with moving parts at another intensity
Maybe I morph erode some layers just slightly

If you go back to layers you've already morphed uvtools doesn't automatically show you the intensity or the pattern you used previously, so I end up needing to take notes. 

khalil nurullah from facebook - the bearing guy

sn4k3, 2020-10-31:
On the Log tab it shows the operations you did in a general way, not in detail.
I will take it into consideration

mylife4aiurr, 2020-11-01:
Ok if possible i'd like uvtools to highlight or shade the model in some visual way on the preview screen changes i've made. I looked at the pixel dimming feature. There is a visual shading/checker pattern applied to model. That helps user see what will happen.

(model view)
I would like something similar for other morph. I'm playing with tolerances to print "print in place" models. I'd like to uvtools to show me the difference in tolerance between current model and morph eroded model (something the dimming features shading), this way I can see how much the tolerance will be increased by or decreased by.

(parameters window)
Can the changes I make to a ranges of layers be remembered by uvtools as a feature enhancement. As I work on different parts on model on make different changes to different ranges of layers. If I return to a range of layers I've worked on previously - the same parameters I used should be set by uvtools.

Thanks for the response, just trying to word my part it better.

sn4k3, 2020-11-01:
> Ok if possible i'd like uvtools to highlight or shade the model in some visual way on the preview screen changes i've made. 

It's possible but that will stack very slow operations, it will be required to subtract images and them append only the changed pixels to current image, that will slow down your image preview alot and operations overall and in the end will not be usable. The only usable method would be create a cached version of image but that will consume 2x more ram to help the speed situation.

> I looked at the pixel dimming feature. There is a visual shading/checker pattern applied to model. That helps user see what will happen.

That's because it's the raw image, same image printer will receive and print. UVtools always show you what printer will display to cure.

> (parameters window)
> Can the changes I make to a ranges of layers be remembered by uvtools as a feature enhancement. As I work on different parts on model on make different changes to different ranges of layers. If I return to a range of layers I've worked on previously - the same parameters I used should be set already by uvtools.

The parameters you previous set MUST be visible on window, uvtools always grab the parameters from layer start index. So if you edit range 20-60, if after you selected any layer from 20 to 60 will show that parameters, BUT if you select layer 19 to 50 for example it will show layer 19 parameters, so uvtools will always grab data from start layer

mylife4aiurr, 2020-11-01:
k

closed 2020-11-01 by mylife4aiurr