sn4k3/UVtools #87: [FEATURE REQUEST] Cant select inverse range

mylife4aiurr, 2020-11-01:
## Is your feature request related to a problem? 
The problem is elephant foot, the layers that get over exposed. i want to start at top layer of bottom layer and apply a given function towards first layer. So 8 to 0, instead of 0 to 8. (example of bottom layer range)

## Describe the solution you'd like
I'd like to start at the highest layer of the bottom layers applying function towards zero layer. Say bottom layers are 0 to 8 - example. To create a chamfer, so thickest layer is 8 and thinest is 0 and then slope is what user chooses in function (morph erode). This cant be done currently, UVtools errors.

sn4k3, 2020-11-01:
I think what you are asking is already implemented, you can do chamfers with the Fade option, try:

[image]

That will erode from 8 to 1

mylife4aiurr, 2020-11-01:
Thank you sir, I will give it a try. Didnt try the fade option.

closed 2020-11-01 by mylife4aiurr