sn4k3/UVtools #88: [FEATURE REQUEST] Elegoo Mars , Mars Pro and Saturn = ChituBox

sandrobilbeisi, 2020-11-01:
## Elegoo 3D resin printers use ChituBox
Elegoo 3D resin printers use Chitubox as their default slicer software , but they are not listed among the supported platforms.

Perhaps the readme should be amended to reflect this.

sandrobilbeisi, 2020-11-01:
The Elegoo version of Chitubox has custom metadata

sn4k3, 2020-11-01:
What do you mean? All that printers are supported by UVtools

sandrobilbeisi, 2020-11-01:
The Elegoo ChituBox export format may be sufficiently unique to warrant a listing among the formats

sn4k3, 2020-11-01:
i still don't get what you are talking about or requesting here. Can you elaborate?

sandrobilbeisi, 2020-11-01:
I apologize for the lack of clarity

I meant to elaborate that the format of Elegoo-Chitubox could be listed in the formats list , such as : 

 👍 Known Formats
SL1 (PrusaSlicer)
Zip (Chitubox)
Photon (Chitubox)
Photons (Chitubox) [Read-only]
CBDDLP (Chitubox)
CBT (Chitubox)
**Elegoo (Chitubox)**
PHZ (Chitubox)
PWS (Photon Workshop)
PW0 (Photon Workshop)
PWMX (Photon Workshop) [Not finished]
ZCodex (Z-Suite)
CWS (NovaMaker)
LGS (Longer Orange 10)
LGS30 (Longer Orange 30)
UVJ (Zip file for manual manipulation format)
Image files (png, jpg, jpeg, gif, bmp)

sn4k3, 2020-11-01:
Elegoo is not a file format

X3msnake, 2020-11-01:
what custom metadata? where are you basing this off?

closed 2020-11-07 by sn4k3