sn4k3/UVtools #90: [BUG] Too long windows in settings with 125% scaled screen

MarkusReichart, 2020-11-03:
In the settings of version 1.1.2 the windows of the tabs "Layer preview" and "Issues" are too long to be fully displayed on a 125% scaled screen.

sn4k3, 2020-11-03:
Zoom overrides are always a problem. Can you send a screenshot?

MarkusReichart, 2020-11-03:
[image]
[image]

Dont be confused. I have a dual screen setup and stared uvtools twice. full screen on each displaz

sn4k3, 2020-11-05:
ok i see the problem, the uvtools windows are height resolution dependent to set the scroll bar and window max height but there are no zoom setting available on this framework to fix this :(

sn4k3, 2020-11-07:
fixed

closed 2020-11-07 by sn4k3