sn4k3/UVtools #92: [BUG] UVtools finds island thats supported

jarjo, 2020-11-04:
## Describe the bug
Indicates that an island that is supported is unsupported

## To Reproduce
Steps to reproduce the behavior:
1. open file
2. Click on issues
4. Island reported on layer 1461 but support visible on 1460 

## Expected behavior
A clear and concise description of what you expected to happen.

## Screenshots
If applicable, add screenshots to help explain your problem. Never hide information, full GUI/window screenshot is always a must!
[image]
[image]

## Files
Were possible attach the problematic file, both source and output file if applicable.
[island_error.zip](https://github.com/sn4k3/UVtools/files/5487403/island_error.zip)

## System
Desktop (please complete the following information):
Win 10 x64 version 10.0.18363

sn4k3, 2020-11-04:
Read this: https://github.com/sn4k3/UVtools/releases/tag/v0.8.6.0

jarjo, 2020-11-04:
that explains it thanks

closed 2020-11-04 by jarjo