sn4k3/UVtools #94: [FEATURE REQUEST] Raise issue when pixels are too close to the edge

nvx, 2020-11-11:
## Is your feature request related to a problem? Please describe
I discovered the hard way that although the screen on an Anycubic Photon is 2560x1440, in practice there is a small margin around the edge where pixels appear to be missing. Making a test file (UV Tools layer import helps a tonne here!) consisting of a blank screen except for a 10px border all around the screen results in a black screen, repeating with a 40px border results in a solid border around the screen, although it is noticeably thinner on the short sides. I'm in the process of testing more thoroughly to work out what the exact margin is for my machine, but I wouldn't be surprised if it varied per screen and/or mobo.

## Describe the solution you'd like
Being able to save a persistent setting in UV Tools for how many pixels on each side (noting the number may be different per side, but likely will be the same for each pair of side, eg same margin on both sides X, and same martin (but different to X) on both sides of Y). It's likely worth having a small out of the box default value too.

The issue would flag up any layer that has illuminated pixels within the safety margin.

## Describe alternatives you've considered
Checking the print boundary box can achieve a similar effect, but is rather tedious.

sn4k3, 2020-11-11:
I have plans to extend the boundary check and be able to set your own margin, right now it's only 1px.
Did you read my comment on fb? about print a outsized object and calculate the offset?

nvx, 2020-11-11:
I did, unfortunately such a wide object is prone to warp or exhibit a little shrinkage in the resin. Instead I printed the attached image (converted into a photon file by UV Tools) - the numbers are too small to read in person, but you can count the larger squares that contain the numbers. I found 30 pixels on the short sides (eg, leaving 2560-60=2500 pixels usable) and 25 pixels on the long sides (eg leaving 1440-50=1390 pixels) is safe. In practice there was a few more pixels beyond that which was safe, but I wasn't going to bother fiddling with less than 5 pixel steps.
[image]

nvx, 2020-11-11:
Considering there's 21.16 pixels per mm, or ~0.047mm per pixel, I doubt I could measure a printed object with more accuracy than simply printing that image and looking at the screen.

sn4k3, 2020-11-11:
ok, i think after you find the best value you should tune slicer LCD Size to give you full print resolution area

nvx, 2020-11-11:
Is that an option in PrusaSlicer? I couldn't see anything like that? The only size I can see is the display width/height which tuning that wouldn't help is my understanding.

sn4k3, 2020-11-11:
You can set lcd size and bed size.
By reduce bed size it will keep you safe from overflow your limit if you tune it right

[image]

nvx, 2020-11-11:
In theory that looks like the right option, but in practice it seems setting that doesn't centre the build plate on the screen but puts it in one corner so you'd still have the issue on one side.

I tried adjusting the origin to be half of the difference between the display size and the print bed size, but it doesn't seem to affect the sliced output file.

This is the output of a sliced file that takes the whole (reduced) built plate:

[image]

And here's the settings:

[image]

Note the file looks the same even if the bed shape origin is set to 0,0

sn4k3, 2020-11-11:
Yeah, that was my guess, non slicer is perfect, chitubox on this setting is better because is centered.
You could use uvtools to center object but less pos-process the better

sn4k3, 2020-11-11:
Implemented, border thickness in all directions

[image]

nvx, 2020-11-12:
Nice, the 5px is configurable?

Unrelated - I noticed my UV Tools has a black background not the checkerboard pattern, is that new or configurable? Or mine broken?

sn4k3, 2020-11-12:
[image]

The checker pattern is disabled by default to speed up layer preview by a few miliseconds, you can enable it on settings under layer preview - Show square grid background

sn4k3, 2020-11-14:
Implemented

closed 2020-11-14 by sn4k3