sn4k3/UVtools #95: [QUESTION]

Aertig, 2020-11-13:
When attempting to run the Resin XP finder tool on an EPAX X1K I encounter several issues. The first if the X1K does not recognize .photon files. After converting the file to .ctb using UVtools, and attempting to run I am presented with the error "different resolution with projector-1440"

While I am not certain, I believe this is related to the EPAX settings within UVtools, listing it as 1440*2560, where as the X1K is 11620*2560. When attempting to reslice the converted .ctb file in Chitubox, the print becomes totally flat and only prints the bottom layer.

Is there a way to change the printer settings in UVtools, or is this something else I am missing?

Thanks.

sn4k3, 2020-11-13:
UVtools doesnt have printers nor adapt to printer, if your input file is 1440x2560, when converting using UVtools the output file will be 1440x2560 untouched. There are Resin XP files per printer i guess, you must use one sliced to your printer or format. 

>Is there a way to change the printer settings in UVtools, or is this something else I am missing?

To match your printer you can use UVtools to change resolution, under tools.

PS: When open a issue here make sure to have a title, so when others come here and search for topics they can know what is this about without open it

Aertig, 2020-11-13:
I just found the option just as you were replying. This fixed the issue I was having and was able to convert and print the test print.

closed 2020-11-13 by sn4k3