sn4k3/UVtools #96: [BUG] UVJ cannot be exported on

yaqwsx, 2020-11-13:
## Describe the bug

When I try to export UVJ as CTB, a message box is shown for a split of a second and then UVTools closes. I wasn't able to obtain the error message as the message box immediately disappeared. Also, no logs were created.

## To Reproduce
Steps to reproduce the behavior:
1. Download [_keycap.zip](https://github.com/sn4k3/UVtools/files/5537970/_keycap.zip) and extract keycap.ujv
2. Open in in UVTools
3. File -> Convert to -> Chitubox
4. Specify a filename and choose CTB
5. Click okay
6. UVTools closes/crashes

## System

 - OS: Windows 10 Pro, x64, 19041.572
 - UVTools 1.2.1

sn4k3, 2020-11-13:
> Convertion was not successful! Maybe not implemented...
> System.NotImplementedException: The method or operation is not implemented.

That means UVJ cant convert back to any other file format due is not implemented yet.
But that's a bug, it should not show the convertion menu when not able to convert, will fix that on next patch and implement the convertion for UVJ

sn4k3, 2020-11-14:
Fixed

closed 2020-11-14 by sn4k3